gazebosim/gazebo-classic #2961: Image viewer not display images correctly for all resolutions

audrow, 2021-04-07:
Setup
* Gazebo 11 from source (bf9608086135866766e7a7d92b85ac2a602761b3)
* Ubuntu Focal in a Docker container

When I use the image viewer to view a camera sensor's output, it seems that the camera image needs to have a width and height that is divisible by four otherwise, I get some strange colors in the image. I assume this is some sort of error indexing the image.

For example, here is the same camera with different image dimensions. Note, this example shows that I'm testing with lens flare enabled, I was also able to see the same error without lens flare. It also occurred with both wide angle and regular cameras.

With 10x10 width and height (not right):
[image]

With 12x12 width and height (looks fine):
[image]

scpeters, 2021-04-14:
can you attach an example world with working and not-working camera resolutions to aid in reproducing this bug?

audrow, 2021-04-15:
> can you attach an example world with working and not-working camera resolutions to aid in reproducing this bug?

Good idea, will do.

audrow, 2021-04-16:
Below is a world that reproduces the bug. I added `.txt` to the file extension so that Github would except the file.

[camera_display_error.world.txt](https://github.com/osrf/gazebo/files/6327327/camera_display_error.world.txt)