gazebosim/gazebo-classic #2964: Bug in BoudingBox() method of class Model

bach05, 2021-04-11:
Hi, 

I think there is a bug in the [model::BoundingBox](https://osrf-distributions.s3.amazonaws.com/gazebo/api/dev/classgazebo_1_1physics_1_1Model.html#a142d84ded73e7ecef9173a8d49b23e84) method. In fact it should return an [ignition::math::Box](https://ignitionrobotics.org/api/math/4.0/classignition_1_1math_1_1Box.html#a4b252bae164ada2860b6140fefee3c9d) described by 2 corners (min and max). 

However I noticed that whenever there is a negative value for the "max" corner, it is rounded to 0. Let's see an example: 

    [Dbg] [AutonomousActorPlugin.cc:224] Collision box: Min[-5 -5.1 0] Max[5 0 2]

In this case the right box would be `Min[-5 -5.1 0] Max[5 -4.9 2]`. It happens every time I think the max corner should have negative values. 

Are there any way to solve the problem? Thank you in advance.

scpeters, 2021-04-14:
Thanks for the report. Can you supply an SDFormat file with a concrete example so we can reproduce the error?

bach05, 2021-04-15:
Of course. This is the world I used in my experiments. It comes from [PAL Robotics Tiago]([url]http://wiki.ros.org/Robots/TIAGo%2B%2B/Tutorials/motions/key_teleop(url)) simulation and it uses a personalized version of [AutonomousActorPlugin](https://github.com/BruceChanJianLe/gazebo-plugin-autonomous-actor). I attach it as **simple_office_with_people_motion2.txt**. 

[simple_office_with_people_motion2.txt](https://github.com/osrf/gazebo/files/6315959/simple_office_with_people_motion2.txt)

You should be able to reproduce the effect by using a simpler world like **simple_office_with_people_motion.txt**. 

[simple_office_with_people_motion.txt](https://github.com/osrf/gazebo/files/6315976/simple_office_with_people_motion.txt)

For example I obtain that wall_3_model has a bounding box `Min[-5.1 -5 0] Max[0 10 2]` that is weird. Don't hesitate to contact me for further questions. For example, if you need it, I can give you the code of my AutonomousActor.