gazebosim/gazebo-classic #2976: Error during Windows installation: CMake will not be able to correctly generate this project

abdo-ameen, 2021-04-20:
Hello,
I'm trying to install Gazebo on my Windows 10 machine and I'm following the tutorial strictly (http://gazebosim.org/tutorials?tut=install_on_windows&cat=install). When I try to configure and build the dependencies I get the following error:

[image]
[image]

any idea how to overcome this error?
many thanks in advance.

traversaro, 2021-04-20:
In this specific case, your environment seems to be compromised, as several compiler/toolchains environment variables seems to be set: 
* You sourced the configuration environment for the **Microsoft Visual Studio 2017** compiler (vcvarsall.bat)
* However, for some reason CMake is finding and using **Microsoft Visual Studio 2019**
* The test compilation of Visual Studio 2019 is failing due to some error of the **MinGW ld** linker.

Do you have any global environment variables (including PATH) that is set related to Visual Studio or MinGW? This may prevent the compilers to work correctly, especially if you mix MinGW and Visual Studio.

However, if you just need to run Gazebo on Windows you can try one of the alternatives methods listed in this issue: https://github.com/osrf/gazebo/issues/2901 . I personally suggest the conda-forge one.

abdo-ameen, 2021-04-21:
Thank you for your comment.
1- The Visual Studio 17/19 mix was resolved, but it didn't change the outcomes.
2- Those are the environment variables I have, no one is related to VS or MinGW
[image]
3- I did the installation using the conda-forge method, I ran the two commands successfully, but then Gazebo could not lunch. I get ("gazebo" is not recognized as internal or external command, operable program or batch file). Basically there are no instructions on how to run Gazebo after conda-forge installation.
4- The only choice left is to fix the CMake issue with "MinGW ld linker". I would appreciate any help in fixing this problem, since I'm not a software developer and this is my first encounter with CMake.

traversaro, 2021-04-21:
>  2- Those are the environment variables I have, no one is related to VS or MinGW

> 4- The only choice left is to fix the CMake issue with "MinGW ld linker". 

Can you print the complete value of the `Path` environment variable. 

> I would appreciate any help in fixing this problem, since I'm not a software developer and this is my first encounter with CMake.

If you are not a software developer, for what do you plan to use Gazebo? Understanding that could help us in suggesting you the best way of installing it.

traversaro, 2021-04-21:
> 3- I did the installation using the conda-forge method, I ran the two commands successfully, but then Gazebo could not lunch. I get ("gazebo" is not recognized as internal or external command, [...]

Are you sure that the environment in which you installed Gazebo as active? Can you report the problem that you are experiencing with Gazebo via conda-forge either in https://github.com/conda-forge/gazebo-feedstock or https://github.com/osrf/gazebo/issues/2976 . Ideally you should report in there the exact environment that you are using, by reporting the output of the commands `conda list` and `conda info`.

abdo-ameen, 2021-04-21:
> 
> Can you print the complete value of the `Path` environment variable.

[image]

> If you are not a software developer, for what do you plan to use Gazebo? Understanding that could help us in suggesting you the best way of installing it.

I'm an engineer and I need it along with ROS for a large part of my master thesis

traversaro, 2021-04-21:
> Can you print the complete value of the `Path` environment variable.

The one related that is causing the `ldd` is `c:\MinGW\bin` . However, all the other paths that contain `MinGW` in the name could create problem as well.

abdo-ameen, 2021-04-21:
I deleted the only two paths that contain `MinGW` and ran `.\configure` again but got similar error
[image]
[image]

traversaro, 2021-04-21:
I think you probably have other problems related to your system being not clean and having a lot of directories in your PATH that contribute to the so-called https://en.wikipedia.org/wiki/DLL_Hell .

> I'm an engineer and I need it along with ROS for a large part of my master thesis

Especially if this is the case and you do not have background in CS and you actually need Gazebo+ROS, I would not focus on compiling Gazebo from source, as in any case then you would need to compile all ROS from source, and I am afraid this is something that is not documented and/or tested, so it may be difficult to achieve. You can look instead at how to install ROS on Windows via the Microsoft official installation (see  http://wiki.ros.org/Installation/Windows) or via conda (see https://github.com/RoboStack/ros-noetic). In both cases Gazebo is included. However notice that with such a "dirty" environment, the risk of experiencing similar problems due to DLL conflicts is quite high. Using a clean Windows or Ubuntu machine (even via Docker/Virtual Machine) may have higher chance of success.

abdo-ameen, 2021-04-21:
Indeed, Gazebo is included with ROS by default!
I installed ROS without problems and now I can launch Gazebo with simple `roslaunch gazebo_ros empty_world.launch` command.
Thank you so much Mr. @traversaro for your time and for keeping up with me :pray: :tulip:

traversaro, 2021-04-21:
You are welcome, @abdo-ameen . If you solved your problem, feel free to close the issue, thanks!

closed 2021-04-21 by abdo-ameen

j-rivero, 2021-04-21:
Thanks Silvio for the help.