gazebosim/gazebo-classic #2984: Gazebo 11.5.0 : [Wrn] [GuiIface.cc:120] Wrong JPEG library version: library is 90, caller expects 62

kevinsmia1939, 2021-04-28:
I create a Flatpak package for Gazebo 11.5.0.

I tried running on openSUSE Tumbleweed KDE.

The Flatpak manifest file with all the dependencies and their version can be found here.
https://github.com/kevinsmia1939/flathub/blob/org.gazebosim.Gazebo/org.gazebosim.Gazebo.yaml

Gazebo Flatpak show splash screen but immediately crash, 

--verbose only give info below.
If you know more way be get error log out of this crash, let me know.

This log is create when running `gazebo --verbose`

But since this is Flatpak, it is,
`flatpak run gazebo --verbose`
```
    Gazebo multi-robot simulator, version 11.5.0
    Copyright (C) 2012 Open Source Robotics Foundation.
    Released under the Apache 2 License.
    http://gazebosim.org

    [Msg] Waiting for master.
    Gazebo multi-robot simulator, version 11.5.0
    Copyright (C) 2012 Open Source Robotics Foundation.
    Released under the Apache 2 License.
    http://gazebosim.org
    [Msg] Waiting for master.
    [Msg] Connected to gazebo master @ http://127.0.0.1:11345
    [Msg] Publicized address: 192.168.2.16
    [Msg] Loading world file [/app/share/gazebo-11/worlds/empty.world]
    [Msg] Connected to gazebo master @ http://127.0.0.1:11345
    [Msg] Publicized address: 192.168.2.16
    [Wrn] [GuiIface.cc:120] Wrong JPEG library version: library is 90, caller expects 62
```

Most likely JPEG library version 90 is given by latest FreeImage 3.18.0. If Gazebo prefer older version let me know.

j-rivero, 2021-04-28:
Thanks @kevinsmia1939 . Gazebo should work with versions >=3.9.0, it surprise me this warning. Could you check if the problem with the jpeg library version is that the binary was built with 62 and now your system has 90?

traversaro, 2021-04-30:
We had this problem in conda/vcpkg as well:  the problem in that case was that FreeImage vendor a libjpeg version itself, so you need either to patch FreeImage to use the external libjpeg library that you use to build Gazebo and the rest of the dependencies (this is what most package managers/distro do) , or at least use the same version of libjpeg that freeimage uses (even if I do not know if this can create further problems if duplicate symbols are present even if ABI compatible, as long as you stick to ELF platforms it should not be a problem).
I am not an expert of Flatpak, but I guess you could have the same problem with any other library that vendors its own dependencies, such as Qt5 unless you configure it with the `-system-libjpeg` flag.

traversaro, 2021-04-30:
For a reference on the patches used to compile FreeImage with an external libjpeg (and also the rest of the dependencies) on Conda/Debian/Fedora:
* https://github.com/conda-forge/freeimage-feedstock/tree/master/recipe
* https://sources.debian.org/patches/freeimage/3.18.0+ds2-6/
* https://src.fedoraproject.org/rpms/freeimage/tree/rawhide

kevinsmia1939, 2021-04-30:
I have a difficult time patching FreeImage at the moment.
There is so many patch, its unreal.

traversaro, 2021-04-30:
I share your pain, whenever I had to port Gazebo to a new package manager FreeImage was always the hardest thing to manage. If you have issue with the autotools build system, you can try to use the CMake one provided in the vcpkg port (see https://github.com/microsoft/vcpkg/tree/master/ports/freeimage), but however there is still a huge amount of patches necessary at the source-code level to de-vendor the used libraries.

kevinsmia1939, 2021-05-02:
> I share your pain, whenever I had to port Gazebo to a new package manager FreeImage was always the hardest thing to manage (see for example [microsoft/vcpkg#3](https://github.com/microsoft/vcpkg/issues/3)). If you have issue with the autotools build system, you can try to use the CMake one provided in the vcpkg port (see https://github.com/microsoft/vcpkg/tree/master/ports/freeimage), but however there is still a huge amount of patches necessary at the source-code level to de-vendor the used libraries.

Hello,

So I have been trying to unbundle FreeImage. Some progress but ran into issues after another.
Latest issue is 

```
Source/FreeImage/PluginJPEG.cpp:638:6: error: ‘jpeg_read_icc_profile’ was not declared in this scope; did you mean ‘jpeg_read_iptc_profile’?
  638 |  if( jpeg_read_icc_profile(cinfo, &icc_profile, &icc_length) ) {
      |      ^~~~~~~~~~~~~~~~~~~~~
      |      jpeg_read_iptc_profile
```

Have you encounter this before?
But anyway, isn't FreeImage retired already?

traversaro, 2021-05-02:
It seems that that function is defined earlier, see https://github.com/gongminmin/FreeImage/blob/7c453922d9442cd1f03a751d1f9273236405f9b7/Source/FreeImage/PluginJPEG.cpp#L506 . Furthermore, it seems provided also by libjpeg-turbo (see https://github.com/libjpeg-turbo/libjpeg-turbo/blob/4de8f6922a9be7d0a51a429e367283fd40031b26/jpeglib.h#L1066) while it is not provided by regular libjpeg (see http://www.ijg.org/files/), so I wonder if you mixed patches to make freeimage work with libjpeg-turbo and then trying linking agains regular libjpeg. Probably it is easier if you could point us to the patches that you are using. 

>  But anyway, isn't FreeImage retired already?

What do you mean?

kevinsmia1939, 2021-05-02:
@traversaro 
Hello,

I manage to unbundle and build FreeImage now.

I use the patch from openSUSE.
https://build.opensuse.org/package/show/openSUSE:Factory/freeimage

bigendian.patch
CVE-2019-12211_2019-12213.patch
doxygen.patch
freeimage_flatpak_dest.patch
freeimage-no-return-in-nonvoid.patch
freeimage-openexr3.patch
libjxr_change_install_path.patch
libraw_0_20.patch
makefiles_fixes.patch
unbundle.patch

With my own freeimage_flatpak_dest.patch which change installation path from /usr/include and /usr/lib to /app/include and /app/lib for Flatpak.

Use compiler flags
```
      cflags: -I/app/include/jxrlib -I/app/include -I/usr/include -fpermissive -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -fPIC -Wunused-variable
      cxxflags: -I/app/include/jxrlib -I/app/include -I/usr/include -fpermissive -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -fPIC -Wunused-variable
```
      
And build it like this.
```
      - rm -rf Source/LibPNG/ Source/LibRawLite/ Source/OpenEXR/ Source/ZLib/ Source/LibOpenJPEG/ Source/LibJPEG/ 
      - truncate -s 0 Source/FreeImageToolkit/JPEGTransform.cpp 
      - truncate -s 0 Source/FreeImage/PluginG3.cpp
      - for file in `find . -type f -name '*.c' -or -name '*.cpp' -or -name '*.h' -or -name '*.txt' -or -name Makefile`; do iconv --from=ISO-8859-15 --to=UTF-8 $file > $file.new && sed -i 's|\r||g' $file.new && touch -r $file $file.new && mv $file.new $file;
        done
      - sh ./gensrclist.sh
      - sh ./genfipsrclist.sh
      - make -f Makefile.gnu
      - make -f Makefile.fip
      - make install
      - make -f Makefile.fip DESTDIR=/app INSTALLDIR=/app/lib install
```

Also libwebp and libjxr must build with -fPIC.

I use libjpeg-turbo instead of libjpeg.

I also working on building testing Flatpak for Gazebo, it does not build successfully at the moment, but if it does I would like someone to test it.
https://github.com/flathub/flathub/pull/2296#issuecomment-830836063 

Thanks.

kevinsmia1939, 2021-05-02:
I will close this now, since this is solve.

closed 2021-05-02 by kevinsmia1939