gazebosim/gazebo-classic #2986: Please help test Flatpak Gazebo build.

kevinsmia1939, 2021-05-02:
This is after I manage to unbundle FreeImage.
https://github.com/osrf/gazebo/issues/2984

I successfully build Gazebo with github CI.
https://github.com/flathub/flathub/pull/2296

Please try it out.
First, install Flatpak. Setup according to your Linux distro.
https://flatpak.org/setup/

Installing the test build by using this command.
`
flatpak install --user https://dl.flathub.org/build-repo/45107/org.gazebosim.Gazebo.flatpakref`

Note that this is not release to Flathub yet.
This install is 500 MB in size because I did not delete any left over development files from its dependencies yet. I will trim it down when ready to ship.

Gazebo flatpak can be run with.
`flatpak run org.gazebosim.Gazebo`

Pass verbose option to Gazebo with something like this.
`flatpak run org.gazebosim.Gazebo --verbose`

Or instead, pass verbose option to Flatpak like this.
`flatpak run -vvv org.gazebosim.Gazebo`

Or access bash in Flatpak like this. And can run Gazebo by typing gazebo in command line inside flatpak.
`flatpak run --command=bash org.gazebosim.Gazebo`

I have issue Gazebo crash on start up. I would like for someone to help test it out.
Thanks.

kevinsmia1939, 2021-05-02:
Update:
So Gazebo show splash screen and interface, but freeze immediately after that.

`flatpak run org.gazebosim.Gazebo --verbose`
Give me this.
Maybe issue with Orge build in Flatpak manifest.

```
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.122.80
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.122.80
[Msg] Loading world file [/app/share/gazebo-11/worlds/empty.world]

(process:5): Gtk-WARNING **: 04:50:48.219: Locale not supported by C library.
	Using the fallback 'C' locale.
[Wrn] [GuiIface.cc:120] Qt: Session management error: Authentication Rejected, reason : None of the authentication protocols specified are supported and host-based authentication failed
[Wrn] [GuiIface.cc:120] Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

[Err] [main.cc:37] Ogre Error:RuntimeAssertionException: Ogre/ShadowExtrudePointLight not found. Verify that you referenced the 'ShadowVolume' folder in your resources.cfg in initialise at /run/build/ORGE/OgreMain/src/OgreShadowVolumeExtrudeProgram.cpp (line 70)
```

kevinsmia1939, 2021-05-05:
I change how I build OGRE now.
I download imgui-1.79 and place it in OGRE folder and build it with imgui now.
However it is still failing with same error.

I build Gazebo like this.
```
      - -DCMAKE_BUILD_TYPE:STRING=Release
      - -DCMAKE_INSTALL_PREFIX:PATH=/app
      - -DBUILD_TESTING:BOOL=ON
      - -DENABLE_PROFILER:BOOL=ON
```

traversaro, 2021-05-05:
Can you try to add `<ogre_install_prefix>/Media` to `GAZEBO_RESOURCE_PATH as mentioned in https://github.com/osrf/gazebo/issues/2700#issuecomment-616319249 ?

kevinsmia1939, 2021-05-05:
@traversaro 
Hi, it is not clear to me what is <ogre_install_prefix>/Media and GAZEBO_RESOURCE_PATH. Is GAZEBO_RESOURCE_PATH a cmake flag?
So something like -DGAZEBO_RESOURCE_PATH=<ogre_install_prefix>/Media
?

traversaro, 2021-05-05:
Sorry, with `<ogre_install_prefix>` I mean the installation path of Ogre that you choose, while `GAZEBO_RESOURCE_PATH` is the environment variable, that is set by setup.sh ( http://gazebosim.org/tutorials?tut=components&cat=get_started ). However, I am not sure how this should work with flatpack. Is Flatpack expecting to deal just with relocatable packages? In that case, you may be interested in https://github.com/osrf/gazebo/issues/2782 .

kevinsmia1939, 2021-05-06:
@traversaro 
Thanks, I did 
`export GAZEBO_RESOURCE_PATH=/app/share/OGRE/Media`
and run gazebo, and the original error is gone.

Can I set this as default somehow?

Gazebo multi-robot simulator, version 11.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

I got new error, looking into it.

```
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.16
[Err] [RTShaderSystem.cc:480] Unable to find shader lib. Shader generating will fail.[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [worlds/empty.world]
[Err] [Server.cc:444] Could not open file[worlds/empty.world]
[Wrn] [Server.cc:359] Falling back on worlds/empty.world
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [worlds/empty.world]
[Err] [Server.cc:444] Could not open file[worlds/empty.world]
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.16
[Err] [RTShaderSystem.cc:480] Unable to find shader lib. Shader generating will fail.
```

traversaro, 2021-05-07:
`/app/share/OGRE/Media` should be appended to `GAZEBO_RESOURCE_PATH`, not overwrite it, otherwise you get that error, see https://github.com/osrf/gazebo/issues/2044 .

kevinsmia1939, 2021-05-21:
Hello @traversaro 

I have been trying to build Gazebo binary and ship it, so everyone can test it out.
I have been busy fixing issue relate to appstream and freeimage, that is why I have not properly test your solution yet.

If you don't mind, try installing this Flatpak build.
You can install Gazebo Flatpak (not submit to Flathub yet).
You need to install Flatpak first. Make sure it is up-to-date. 
I assume that it is Ubuntu, Ubuntu ship old version of Flatpak, need to update first.
`https://flatpak.org/setup/Ubuntu/`

Now you can install Gazebo with this command.
`flatpak install --user https://dl.flathub.org/build-repo/46914/org.gazebosim.Gazebo.flatpakref`

There should be a desktop icon there as well, but I recommend using the command for now. Because Gazebo freeze.

`flatpak run org.gazebosim.Gazebo --verbose`
will give the error below.

```
Gazebo multi-robot simulator, version 11.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.16
[Msg] Loading world file [/app/share/gazebo-11/worlds/empty.world]
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.16
[Wrn] [GuiIface.cc:120] Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

[Err] [main.cc:37] Ogre Error:RuntimeAssertionException: Ogre/ShadowExtrudePointLight not found. Verify that you referenced the 'ShadowVolume' folder in your resources.cfg in initialise at /run/build/OGRE/OgreMain/src/OgreShadowVolumeExtrudeProgram.cpp (line 70)
```

So I tried your solution.
Using this command `--command=bash` to run bash inside Gazebo build.
Then export env.
```
kev@localhost:~/flathub> flatpak run --command=bash org.gazebosim.Gazebo
[📦 org.gazebosim.Gazebo flathub]$ export GAZEBO_RESOURCE_PATH=${GAZEBO_RESOURCE_PATH}:/app/share/OGRE/Media
[📦 org.gazebosim.Gazebo flathub]$ gazebo --verbose
Gazebo multi-robot simulator, version 11.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.16
[Err] [RTShaderSystem.cc:480] Unable to find shader lib. Shader generating will fail.[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [worlds/empty.world]
[Err] [Server.cc:444] Could not open file[worlds/empty.world]
[Wrn] [Server.cc:359] Falling back on worlds/empty.world
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [worlds/empty.world]
[Err] [Server.cc:444] Could not open file[worlds/empty.world]
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.16
[Err] [RTShaderSystem.cc:480] Unable to find shader lib. Shader generating will fail.
```

But I got the error above.
Any idea?

Thanks.

traversaro, 2021-05-21:
I think you need to run the setup.sh of Gazebo (not sure where it is installed with flatpack) before setting `GAZEBO_RESOURCE_PATH`, see https://github.com/osrf/gazebo/issues/2044 .

kevinsmia1939, 2021-05-22:
> I think you need to run the setup.sh of Gazebo (not sure where it is installed with flatpack) before setting `GAZEBO_RESOURCE_PATH`, see #2044 .

Hello @traversaro 
The setup.sh is locate in /app/share/gazebo/setup.sh
With latest build, I copy it to /app/bin/setup.sh
I run Gazebo flatpak like this.

Install latest Gazebo flatpak build.
`flatpak install --user https://dl.flathub.org/build-repo/46958/org.gazebosim.Gazebo.flatpakref`
from here: https://github.com/flathub/flathub/pull/2296

Run bash inside Gazebo flatpak build.
`flatpak run --command=bash org.gazebosim.Gazebo`

Check if setup.sh is present.
`cat /app/bin/setup.sh` which give output below.
```
export GAZEBO_MASTER_URI=${GAZEBO_MASTER_URI:-http://localhost:11345}
export GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
export GAZEBO_RESOURCE_PATH=/app/share/gazebo-11:${GAZEBO_RESOURCE_PATH}
export GAZEBO_PLUGIN_PATH=/app/lib/gazebo-11/plugins:${GAZEBO_PLUGIN_PATH}
export GAZEBO_MODEL_PATH=/app/share/gazebo-11/models:${GAZEBO_MODEL_PATH}
export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/app/lib/gazebo-11/plugins
export OGRE_RESOURCE_PATH=/app/lib/OGRE
```

Run setup.sh
`setup.sh`

Run Gazebo.
`gazebo --verbose`
Got this, UI appear but freeze.
```
Gazebo multi-robot simulator, version 11.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.12
[Msg] Loading world file [/app/share/gazebo-11/worlds/empty.world]
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.12
[Wrn] [GuiIface.cc:120] Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

[Err] [main.cc:37] Ogre Error:RuntimeAssertionException: Ogre/ShadowExtrudePointLight not found. Verify that you referenced the 'ShadowVolume' folder in your resources.cfg in initialise at /run/build/OGRE/OgreMain/src/OgreShadowVolumeExtrudeProgram.cpp (line 70)
```

Try again
`setup.sh && export GAZEBO_RESOURCE_PATH=${GAZEBO_RESOURCE_PATH}:/app/share/OGRE/Media`

Check environment with `env`.
```
SHELL=/bin/sh
SESSION_MANAGER=local/localhost:@/tmp/.ICE-unix/1400,unix/localhost:/tmp/.ICE-unix/1400
WINDOWID=0
COLORTERM=truecolor
XDG_CONFIG_DIRS=/app/etc/xdg:/etc/xdg
LESS=-M -I -R
[... 110 lines cut ...]
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.12
[Err] [RTShaderSystem.cc:480] Unable to find shader lib. Shader generating will fail.[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [worlds/empty.world]
[Err] [Server.cc:444] Could not open file[worlds/empty.world]
[Wrn] [Server.cc:359] Falling back on worlds/empty.world
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [worlds/empty.world]
[Err] [Server.cc:444] Could not open file[worlds/empty.world]
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.12
[Err] [RTShaderSystem.cc:480] Unable to find shader lib. Shader generating will fail.
```

Any idea?
Thank.

kevinsmia1939, 2021-05-29:
Update
I run
`export GAZEBO_RESOURCE_PATH=${GAZEBO_RESOURCE_PATH}:/app/share/OGRE/Media/ShadowVolume`
 in addition to setup.sh and `export GAZEBO_RESOURCE_PATH=${GAZEBO_RESOURCE_PATH}:/app/share/OGRE/Media`

Now the error is,

```
Gazebo multi-robot simulator, version 11.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.14
[Msg] Loading world file [/app/share/gazebo-11/worlds/empty.world]
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.14
[Wrn] [GuiIface.cc:120] Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

[Err] [main.cc:37] Ogre Error:RuntimeAssertionException: gpu program could not be created in createGpuPrograms at /run/build/OGRE/Components/RTShaderSystem/src/OgreShaderProgramManager.cpp (line 251)
```

kevinsmia1939, 2021-05-29:
Update
From the error,
` Ogre Error:RuntimeAssertionException: gpu program could not be created in createGpuPrograms at /run/build/OGRE/Components/RTShaderSystem/src/OgreShaderProgramManager.cpp (line 251)`

I found this thread.
`https://forums.ogre3d.org/viewtopic.php?t=95581`

Which suggest adding `return true;` to line 108 in ` ogre/Components/RTShaderSystem/src/OgreShaderProgramProcessor.cpp`

So I create patch to OGRE 1.12.12.

```
diff -ruN a/Components/RTShaderSystem/src/OgreShaderProgramProcessor.cpp b/Components/RTShaderSystem/src/OgreShaderProgramProcessor.cpp
--- a/Components/RTShaderSystem/src/OgreShaderProgramProcessor.cpp	2021-04-18 06:55:39.000000000 +0700
+++ b/Components/RTShaderSystem/src/OgreShaderProgramProcessor.cpp	2021-05-29 17:07:32.687497070 +0700
@@ -105,7 +105,7 @@
 //-----------------------------------------------------------------------------
 bool ProgramProcessor::compactVsOutputs(Function* vsMain, Function* fsMain)
 {
-
+return true;
     int outTexCoordSlots;
     int outTexCoordFloats;
```

but still same error when I rebuild Gazebo and OGRE.
`[Err] [main.cc:37] Ogre Error:RuntimeAssertionException: gpu program could not be created in createGpuPrograms at /run/build/OGRE/Components/RTShaderSystem/src/OgreShaderProgramManager.cpp (line 251)`

Tobias-Fischer, 2021-07-20:
Hi @kevinsmia1939, did you have any more luck in debugging this?

kevinsmia1939, 2021-07-20:
> Hi @kevinsmia1939, did you have any more luck in debugging this?

No luck, my debugging skill is very limited.

kevinsmia1939, 2021-08-04:
Hi,

I use Ogre 1.9.0 and Gazebo 11.7.0, Gazebo does not show
Ogre Error:RuntimeAssertionException: gpu program could not be created in createGpuPrograms at /run/build/OGRE/Components/RTShaderSystem/src/OgreShaderProgramManager.cpp error anymore.

Now Gazebo is working and does not freeze now.

I can keep trying with newer Orge to see which version broke.

kevinsmia1939, 2021-08-05:
Ogre 
1.9.0 work
1.10.12 work
1.11.6 broke
`[Err] [main.cc:37] Ogre Error:InternalErrorException: Not all parameters could be constructed for the sub-render state. in IntegratedPSSM3::resolveParameters at /run/build/Gazebo/gazebo/rendering/CustomPSSMShadowCameraSetup.cc (line 236)`
1.12.9 broke
`Ogre Error:RuntimeAssertionException: gpu program could not be created in createGpuPrograms at /run/build/OGRE/Components/RTShaderSystem/src/OgreShaderProgramManager.cpp (line 251)`
1.12.12 broke
`[Err] [main.cc:37] Ogre Error:RuntimeAssertionException: gpu program could not be created in createGpuPrograms at /run/build/OGRE/Components/RTShaderSystem/src/OgreShaderProgramManager.cpp (line 251)`

kevinsmia1939, 2021-08-06:
Do I need to ship Gazebo with desktop icon?

I include Player driver with Gazebo, do Gazebo support all the driver that Player offer?

Do I need to also ship OpenCV?

Tobias-Fischer, 2021-08-07:
Looking at the file that throws the error and the diff between 1.11.3 and 1.11.4, it seems like this PR might be the culprit: https://github.com/OGRECave/ogre/pull/930

@kevinsmia1939 - if you still have energy, I would suggest a build just before this PR (https://github.com/OGRECave/ogre/commit/76aaac47a1910ac846f3ff9519cf478f87863ad4) and just after (https://github.com/OGRECave/ogre/commit/c9ac8b00e12df2ba2a574ecca18d21ad9b4d5f82). 

Disclaimer: I have no real clue what's going on in this PR, and am just using intuition..

Tobias-Fischer, 2021-08-07:
And it seems like the error changes from 1.11 to 1.12 as well, so there well might be more than one issue.

kevinsmia1939, 2021-08-07:
@Tobias-Fischer 
I just build ogre 1.12.0 with gazebo 11.7.0 and it work without the issue now, probably was fixed.
Im trying ogre 1.12.5  atm.

traversaro, 2021-08-07:
> 
> 
> Do I need to ship Gazebo with desktop icon?
> 
> I include Player driver with Gazebo, do Gazebo support all the driver that Player offer?
> 
> Do I need to also ship OpenCV?

I think it really depends which use cases are you targeting with this Flatpak Gazebo build. Many users use Gazebo simulation by writing software using robotics middlewares such as ROS or ROS2, or build their own C++ plugins, but I am not sure how this would work with Flatpak or if this is the intended use case of Flatpak.

traversaro, 2021-08-07:
> 
> 
> @Tobias-Fischer
> I just build ogre 1.12.0 with gazebo 11.7.0 and it work without the issue now, probably was fixed.
> Im trying ogre 1.12.5 atm.

Great, thanks a lot for bisecting!