gazebosim/gazebo-classic #2987: Foxy/Gazebo11 does not load URDF model with meshes, from robot_descritpion

newcanopies, 2021-05-03:
Following the http://gazebosim.org/tutorials?tut=ros_roslaunch&cat=connect_ros tutorial method to spawn URDF from  "Model Database" - DOES NOT WORK in Gazebo11 ROS2 Foxy

Gazebo loads the exported Model Database path in the Insert tab, but the robot is not listed on that path. 

_FEATURE REQUEST_
it would be great to have a ROS2 version of this command to quickly verify URDF in Gazebo11

`rosrun gazebo_ros spawn_model -file `rospack find MYROBOT_description`/urdf/MYROBOT.urdf -urdf -x 0 -y 0 -z 1 -model MYROBOT`

j-rivero, 2021-05-03:
I think that the issue belongs more to the repo https://github.com/ros-simulation/gazebo_ros_pkgs than to Gazebo itself. Not sure if they can help you but, did you read the [migration notes on spawn/delete for ROS2](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Spawn-and-delete)?

If that does not help you at all, please open an issue in the `gazebo_ros_pkgs` repository.
Thanks!

closed 2021-05-03 by j-rivero

newcanopies, 2021-05-03:
thanks @j-rivero, I read/tested ALL the available documentation including [ migration notes on spawn/delete for ROS2](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Spawn-and-delete), but **none of the available methods** load a URDF with visual meshes!

newcanopies, 2021-05-03:
https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1236

newcanopies, 2021-05-03:
Same issue:
https://answers.gazebosim.org/question/26073/cannot-spawn-urdf-into-gazebo-using-ros2/

Possible solution:
https://answers.ros.org/question/373495/ros2-foxy-gazebo-spawn_entity-systempathscc459-file-or-path-does-not-exist/