gazebosim/gazebo-classic #2996: Issue 868 again: Setting position on heightmap creates offset between collision and visual

Darkproduct, 2021-05-06:
I have the exact same problem on Gazebo 11.3.0. Changing the heightmap position to 0,0 fixes the missmatch between the collision and the visual mesh. This is hard to debug, because the heightmap's collision mesh is not visible even when enabeling collision visualization in the menu. 

[Here](https://answers.gazebosim.org//question/4555/setting-position-on-heightmap-creates-offset-between-collision-and-visual/) is the old discussion, on answers.gazebosim.org.

```
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org
```

Darkproduct, 2021-05-13:
It turns out the heightmap caching wasn't the issue. But there are some problems as well.

flabrosse, 2022-01-07:
Same problem here.  Using the 'Winding Valley Heightmap' model, I changed the height of the visual and collision to 0 (from -4)  and all is fine.  If I shift both along x or y by the same amount, the visual is moved but it looks like the collision isn't.  So a vertical translation is OK but not a horizontal one.

Yangpingz, 2022-10-24:
Same problem happened to me