gazebosim/gazebo-classic #3000: "gazebo::physics::Joint" has no member named "GetAngle"

RainySun-GT, 2021-05-08:
Hi, I'm using gazebo9.

During my developement of other package, it seems that the package depends on the earlier version of gazebo and cmake error occurs and show that " error: **‘class gazebo::physics::Joint’ has no member named ‘GetAngle’**"

I have checked history version of the header file "Joint.hh" in which the definition of the class gazebo::physics::Joint is given. And I find that the member method "GetAngle()" has been  deprecated after gazebo7.

For example, in the branch of "gazebo7-7.0.0", header file "Joint.hh" gives the definition of "GetAngle()" as follow:
'''
public: math::Angle GetAngle(unsigned int _index) const;
'''

I wonder is there any available member method in the new version header file "Joint.hh" that has the same function with the deprecated method GetAngle() ?

I'm looking forward to your reply! Thank you!

traversaro, 2021-05-08:
I think you can find this info in the Migration.md file: https://github.com/osrf/gazebo/blob/gazebo11_11.5.1/Migration.md . In particular in your case:
> Deprecation: math::Angle GetAngle(unsigned int _index) const
Replacement: virtual double Position(const unsigned int _index = 0) const final

RainySun-GT, 2021-05-10:
Thanks a lot for your reminder of the file which helps me fix some errors
And one more thing about the "ConnectionPtr Events::DiconnectWorldUpdateBegin(T _subscriber)" in the Events.hh file, which seems also an deprecation in newer version, but i can't find the replacement info in the Migration.md file.
Do you have any other references ? Sorry to bother you.