gazebosim/gazebo-classic #3001: Problem installing libgazebo9-dev in Ubuntu 18.04 using apt packages from http://packages.osrfoundation.org/ repo

traversaro, 2021-05-10:
Since May 7 2021, some nightly jobs that install `libgazebo9-dev` on Ubuntu 18.04 (see https://github.com/robotology/gazebo-yarp-plugins/actions) are failing with error:
~~~
2021-05-10T02:44:44.7709359Z Some packages could not be installed. This may mean that you have
2021-05-10T02:44:44.7710732Z requested an impossible situation or if you are using the unstable
2021-05-10T02:44:44.7711756Z distribution that some required packages have not yet been created
2021-05-10T02:44:44.7712600Z or been moved out of Incoming.
2021-05-10T02:44:44.7713402Z The following information may help to resolve the situation:
2021-05-10T02:44:44.7714449Z 
2021-05-10T02:44:44.7715210Z The following packages have unmet dependencies:
2021-05-10T02:44:44.8340549Z  libgazebo9-dev : Depends: libignition-transport4-dev but it is not going to be installed
2021-05-10T02:44:44.8342336Z                   Depends: libignition-fuel-tools1-dev but it is not going to be installed
2021-05-10T02:44:44.8724818Z E: Unable to correct problems, you have held broken packages.
2021-05-10T02:44:44.8745707Z ##[error]Process completed with exit code 100.
~~~

I thought at the beginning this was related to https://github.com/osrf/gazebo/issues/2995, but it turns out that the problem is persistent.

klaxalk, 2021-05-13:
I second this issue.

j-rivero, 2021-05-13:
CI is fine ([image]) with new systems having libgazeb9-dev from `packages.osrfoundation.org`. CI seems also fine with having `libgazebo9-dev` on Xenial with `packages.osrfoundation.org` and `packages.ros.org` ([image]).

Adding `-o Debug::pkgProblemResolver=yes` to your apt invocations should give us some light about what's going on.

traversaro, 2021-05-13:
I am not sure what changed, but while the issue was still there in the nightly jobs ~10 hours ago (see https://github.com/robotology/gazebo-yarp-plugins/runs/2572047842?check_suite_focus=true), while now the issue is solved (see https://github.com/robotology/gazebo-yarp-plugins/runs/2575607040?check_suite_focus=true).

traversaro, 2021-05-13:
Related comment https://github.com/robotology/robotology-superbuild/issues/715#issuecomment-840680160 : 
> This problem disappeared. However, in the case of the robotology-superbuild CI we did not use the http://packages.osrfoundation.org/ apt packages, so the whole assumption behind opening osrf/gazebo#3001 was wrong. Given that by looking at https://github.com/robotology/robotology-superbuild/actions/runs/818895297 it seems that the build on the vanilla Docker bionic images were not affected by the problem, while the builds on the ubuntu-18.04 image of GitHub Actions were affected, probably the issue was caused by some strange problem in the GitHub Action ubuntu-18.04 image.

In a nutshell, I do not think there was any problem in the http://packages.osrfoundation.org/, but the issue was indeed due to some strange GitHub Actions problem. In any case, I think we can close it, @klaxalk feel free to comment if instead you are still experiencing the problem.

@j-rivero thanks for the feedback and sorry for the noise!

closed 2021-05-13 by traversaro

j-rivero, 2021-05-13:
> @j-rivero thanks for the feedback and sorry for the noise!

you are welcome, please reopen if somehow the problem is still there.

klaxalk, 2021-05-14:
Thanks to you all and to @traversaro, the problem seems to be fixed now!