gazebosim/gazebo-classic #3002: Collada Material not correctly loaded when using Ogre >= 1.10

traversaro, 2021-05-10:
Original conda-forge issue: https://github.com/conda-forge/gazebo-feedstock/issues/74 .

This issue was reported (with a lot of useful detail, thanks!) by @danzimmerman in https://github.com/conda-forge/gazebo-feedstock/issues/74, but it turns out that is an upstream issue, that is tipically not noticed as most official binaries of Gazebo use Ogre 1.9, and the problem is only present with Ogre >= 1.10 . 

The issue is fixed with the patch https://github.com/conda-forge/gazebo-feedstock/blob/6f01a921c3bd994461dac835b7d8d54bc280e6f5/recipe/fix-invisible-meshes.patch on both ROSOnWindows and conda-forge's Gazebo binaries, but I am not sure if such a patch would be welcome upstream (given the use of static variables, or if anyone has any idea for a cleaner solution.

iche033, 2021-05-10:
I think the static variable is fine. One concern is when users repeatedly remove and insert the model, a new material would be created each time. Maybe it's worth checking to see if this causes increase in memory usage, if so, a line can be added to remove the existing material before creating the new one to avoid memory leaks.

traversaro, 2021-05-19:
> I think the static variable is fine. One concern is when users repeatedly remove and insert the model, a new material would be created each time. Maybe it's worth checking to see if this causes [...]

Yes, that seems to be the case. Probably the root issue is the same that (for ignition-rendering) is solved by https://github.com/ignitionrobotics/ign-rendering/pull/324 .

traversaro, 2021-05-20:
Possibly other related issue: https://github.com/ignitionrobotics/ign-gui/issues/208 .