gazebosim/gazebo-classic #3003: Distortion implementation correction

kjeppesen1, 2021-05-12:
When undistorting a distorted image taken in Gazebo, the focal length must be specified as the image resolution for it to be correct.

I believe the issue is in the "normalization" of the image pixels that are put into Brown's Distortion Equations. I have attached some slides explaining the fix, including some tests performed to demonstrate it. The short description is that the distortion must occur in the camera plane rather than the image plane, and thus the pixel coordinates must first be projected to the camera plane via dividing by the focal length. Then after distortion the focal length is then multiplied to project the image back to the image plane.
[Gazebo Distortion Modification.pdf](https://github.com/osrf/gazebo/files/6468096/Gazebo.Distortion.Modification.pdf)

The relevant function is [Distortion::Distort()](https://github.com/osrf/gazebo/blob/e2436b3a2b32713e2b069f23689b62786273d773/gazebo/rendering/Distortion.cc#L385).

There is a minor change to the math performed at the distortion, and there is a need for Distortion.cc to have access to the focal length parameter (not implemented here). Here is a fix for the function itself that I tested with:
`ignition::math::Vector2d Distortion::Distort2(
    const ignition::math::Vector2d &_in,
    const ignition::math::Vector2d &_center, double _k1, double _k2, double _k3,
    double _p1, double _p2, unsigned int _width, double _f)
{
  // apply Brown's distortion model, see
  // http://en.wikipedia.org/wiki/Distortion_%28optics%29#Software_correction

  ignition::math::Vector2d normalized2d = ((_in*_width) - (_center*_width))/_f;
  ignition::math::Vector3d normalized(normalized2d.X(), normalized2d.Y(), 0);
  double rSq = normalized.X() * normalized.X() +
               normalized.Y() * normalized.Y();

  // radial
  ignition::math::Vector3d dist = normalized * (1.0 +
      _k1 * rSq +
      _k2 * rSq * rSq +
      _k3 * rSq * rSq * rSq);

  // tangential
  dist.X() += _p2 * (rSq + 2 * (normalized.X()*normalized.X())) +
      2 * _p1 * normalized.X() * normalized.Y();
  dist.Y() += _p1 * (rSq + 2 * (normalized.Y()*normalized.Y())) +
      2 * _p2 * normalized.X() * normalized.Y();

  return ((_center*_width) + ignition::math::Vector2d(dist.X(), dist.Y())*_f)/_width;
}`

scpeters, 2021-05-12:
In terms of passing the focal length to `Distortion.cc`, here's some discussion of how these parameters are defined in SDFormat. A `<camera>` element can hold `<`[distortion](https://github.com/osrf/sdformat/blob/sdformat9_9.5.0/sdf/1.6/camera.sdf#L78-L98)`>` and `<`[lens](https://github.com/osrf/sdformat/blob/sdformat9_9.5.0/sdf/1.6/camera.sdf#L100-L155)`>` elements. I believe the focal length parameters are defined in [//lens/intrinsics](https://github.com/osrf/sdformat/blob/sdformat9_9.5.0/sdf/1.6/camera.sdf#L137-L154).

In [gazebo::rendering::Camera::Load](https://github.com/osrf/gazebo/blob/gazebo11_11.5.1/gazebo/rendering/Camera.cc#L184-L210), the distortion parameters are loaded first, and then the lens intrinsics are loaded directly after. It looks like the focal length parameters are not stored, but used to [build a new projection matrix](https://github.com/osrf/gazebo/blob/gazebo11_11.5.1/gazebo/rendering/Camera.cc#L212-L241).

scpeters, 2021-05-12:
It sounds like the field of view parameters are relevant as well and always available. It would be worth trying to access those parameters from within the [Distortion::SetCamera](https://github.com/osrf/gazebo/blob/gazebo11_11.5.1/gazebo/rendering/Distortion.cc#L133) function. You can call [Camera::HFOV](https://github.com/osrf/gazebo/blob/gazebo11_11.5.1/gazebo/rendering/Camera.cc#L851-L855) using the camera pointer in that function to get the horizontal field of view angle.

kjeppesen1, 2021-05-13:
Here is the working implementation that also grabs the FOV: https://github.com/kjeppesen1/gazebo/tree/distortion_normalization

The resulting image gets a bit weird when using extreme distortions with large FOVs, whereas the default implementation does not see that happen.

kjeppesen1, 2021-05-22:
Audrow and I discovered that the strange behavior for extreme distortions noted above was indeed the image wrapping around itself, and the reason it only occurred on the lower half of the image is because the values get over-written from top to bottom. The fix is to check if multiple values are being assigned to the same pixel location and check which one is less offset from the center of distortion.

See here for a comparison for when k1 = -0.8, HFOV = 1.2217 (70deg) with and without this described patch:
[image] 
[image]

kjeppesen1, 2021-05-22:
New draft PR for this is here: https://github.com/osrf/gazebo/pull/3009

kjeppesen1, 2021-06-04:
See PR #3009 for a discussion of not doing a border crop at all with this new implementation. However for completeness here is a comparison of using the current border crop method (uses the old Distortion::Distort method) vs using a new one (uses the new Distortion::Distort method), with the image itself being distorted with the new method and the scaling factor being calculated with either the old or new.

k1 = -0.2 with no cropping
[image]

k1 = -0.2 cropping with old Distortion::Distort()
[image]

k1=-0.2 cropping with new Distortion::Distort()
[image]

k1=-0.8 with no cropping
[image]

k1=-0.8 cropping with old Distortion::Distort()
[image]

k1=-0.8 cropping with new Distortion::Distort()
[image]

Using the old distort method to calculate the scaling factor for border cropping still leaves a black border. Using the new distort method for this correctly cuts away the black border, but for extreme distortions that would result in wrapping as discussed above, the cropping is also extreme. This is supplementary information if it is decided to do no cropping at all in PR #3009.

PR #3009 (2021-06-21): Distortion normalization improvement + fix "folding" on large distortions
Fixes #3003.

Modifies how Brown's distortion equations are applied to better reflect real distortion. Image is projected from image plane to camera plane to apply distortion equations, then projected back to image plane.

Before/after focal scaling implementation:
[image]

Before/after extreme distortion edge case fix:
[image] [image]

---

We use a tag `<ignition:legacy_mode>` in a distortion context to select this new behavior.

Regardless of if the new mode or legacy mode are used, this PR adds a bug fix for heavy di [... cut]
closed 2021-06-21 by scpeters