gazebosim/gazebo-classic #3010: Windows CMake error "Could NOT find Protobuf"

jacobperron, 2021-05-24:
The last two commits (e2436b3a2b32713e2b069f23689b62786273d773 and ce8b0ed68908e59059b1dbd97e8187d5e7640778) have failed to build on Windows due to the following error:

```
CMake Error at C:/Program Files/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Protobuf (missing: Protobuf_LIBRARIES Protobuf_INCLUDE_DIR)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.20/Modules/FindProtobuf.cmake:646 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/SearchForStuff.cmake:34 (find_package)
  CMakeLists.txt:163 (include)
```

First occurrence: https://build.osrfoundation.org/job/gazebo-ci-gazebo11-windows7-amd64/65/

https://github.com/osrf/gazebo/pull/2897 touches some CMake code in the tests, but it looks unrelated as the failure originates from a section before the modified code.

jacobperron, 2021-05-24:
Looks like this is also affecting Gazebo 9: https://build.osrfoundation.org/job/gazebo-ci-gazebo9-windows7-amd64/202

@j-rivero Could this be provisioning issue related to the Windows hosts?

j-rivero, 2021-05-25:
> @j-rivero Could this be provisioning issue related to the Windows hosts?

It is. In fact the problem comes from the lack of wget on the Windows PATH. I added an explicit message for failures of that type in https://build.osrfoundation.org/job/gazebo-ci-gazebo11-windows7-amd64/68/. I'm unable to make batch processing to behave correctly under errors in `call` to make it fail at the right point of the build.

I've fixed the problem in Beefy, back to action [image]