gazebosim/gazebo-classic #3023: update from 9.16 to 9.19 with binary: Assertion `pRep' failed

Wayne-xixi, 2021-06-14:
Hello,

I updated Gazebo from 9.16 to 9.19  today and a new problem appeared when I launched the same simulation as before. 

`gzserver: /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreSharedPtr.h:253: T* Ogre::SharedPtr<T>::operator->() const [with T = Ogre::ShadowCameraSetup]: Assertion `pRep' failed.`

Any help would be greatly appreciated.

j-rivero, 2021-06-14:
Probably the easiest thing to do is to check the [Changelog](https://github.com/osrf/gazebo/blob/gazebo9/Changelog.md) for possible related changes affecting your user case, seems like 9.6.17 had some changes related to Camera and Ogre. It would be great if the exact PR where the problem is triggered is detected.

In parallel, having a minimal case for reproducing the error would be great, so we can look into the problem. As a workarond, you remove latest version and install an old versions of gazebo9 [downloading .debs for 9.6.16](http://gazebosim.org/distributions/gazebo/releases/) and using dpkg -i to install all them.

Wayne-xixi, 2021-06-14:
> Probably the easiest thing to do is to check the [Changelog](https://github.com/osrf/gazebo/blob/gazebo9/Changelog.md) for possible related changes affecting your user case, seems like 9.6.17 had [...]

Hi, thank you very much for quick reply. I have made a few tests to locate the error. It has to be proved that, the error came from  the "cast_shadows=true"  in sdf. As below is the description I used in sdf in gazebo 9.16. But in 9.19 it didn't work. If I set false, and then gazebo 9.19 works fine without error. 
`   <cast_shadows>true</cast_shadows> `

Hope it is useful infos for you. ANd many thanks for your help.

j-rivero, 2021-06-14:
> Hope it is useful infos for you. ANd many thanks for your help.

Sure it helps it is a very good hint to solve the issue. Can you attach/incldue a small demo world file that triggers the error?

Wayne-xixi, 2021-06-14:
> > Hope it is useful infos for you. ANd many thanks for your help.
> 
> Sure it helps it is a very good hint to solve the issue. Can you attach/incldue a small demo world file that triggers the error?

pls see attachment.
[empty_world_3023.txt](https://github.com/osrf/gazebo/files/6650904/empty_world_3023.txt)

j-rivero, 2021-06-14:
thanks, I can not reproduce the problem, seems to work fine for me. gazebo 9.19.0 version on Ubuntu Bionic. Can you please give a look into the logs `$HOME/.gazebo` (specially ~/.gazebo/ogre.log) to see if there is anything reported there?

Wayne-xixi, 2021-06-14:
> thanks, I can not reproduce the problem, seems to work fine for me. gazebo 9.19.0 version on Ubuntu Bionic. Can you please give a look into the logs `$HOME/.gazebo` (specially ~/.gazebo/ogre.log) to see if there is anything reported there?

[empty_world_3023.txt](https://github.com/osrf/gazebo/files/6651508/empty_world_3023.txt)
Sorry, please try this one. The earlier one was something wrong.

rinnaz, 2021-07-26:
It seems that `<cast_shadows>true</cast_shadows>` from `<light>` element conflicts with the scene setting:

```
    <scene>
      <shadows>0</shadows>
    </scene>
```