gazebosim/gazebo-classic #3025: Visual Markers not showing up in sensors (Camera)

nocoinman, 2021-06-14:
### Platform

Gazebo version 11.6
ROS Noetic

### Description

Followed [these](https://github.com/osrf/gazebo/tree/gazebo11/examples/stand_alone/marker) examples for adding visual markers into gazebo but these markers are not picked up by sensors like cameras. I tried setting the visibility to all via 
```cpp
 _markerMsg.set_visibility(ignition::msgs::Marker_Visibility_ALL);
```
but this was of no help

nocoinman, 2021-06-17:
From [these lines](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/rendering/Scene.cc#L417) I see that the marker manager isn't initialised on the server side, thus making the visual markers invicible to cameras. Is there any way to make visual markers visible to cameras? I wanted to use this feature to simulate markings on the ground plane and control them dynamically during runtime via ROS.

iche033, 2021-06-17:
how about instead of using markers, just spawn static models with only visuals and no collisions?

nocoinman, 2021-06-17:
I am intending to use this feature to simulate a cleaning task. You have a patch on the ground, say a square, robot attempts to clean it and when it does I want only that section of the patch to disappear. My idea was to divide the patch into small triangles and use the `TRIANGLE_LIST` markers, which will be deleted on cleaning.

I don't think we can clear parts of a model's visual. A naive approach would be to use a static model with only visual for each of the smaller triangles, but this will be too heavy when I need to increase the resolution and number of patches.

Another attempt at this was a visual plugin that I attached to the ground, and make use of the `CreateDynamicLine()` method of the `rendering::Visual` as shown in this [answer](https://answers.gazebosim.org/question/3383/how-to-add-a-dynamic-visual-marker-in-gazebo/?answer=3394#post-id-3394). I changed the `RenderOpType` to `RENDERING_TRIANGLE_LIST` and set the visibility flags to `GZ_VISIBILITY_ALL` to make the visual visible in the camera. But the behaviour of this was quite strange. Gazebo either rendered it in the GUI or in the camera image, but not both, and this was random on different runs as well.

iche033, 2021-06-18:
hmm I don't know why it's only visible in either the server or the client. You can try adding a print statement in the `Load` function to see if it gets printed out twice. If you only see the printout once, then that indicates the issue is due to the visual plugin only being loaded on one side. If it gets printed out twice, then the issue could be in the rendering system.

nocoinman, 2021-06-19:
@iche033 Fixed it!! The fact that the plugin gets loaded on both the client side and server side didn't click with me initially. My visual plugin initialised a ROS server internally to listen for services and since the plugin got loaded on both sides, whichever side initialised the ROS server first got triggered and rendered the patch. By using subscribers instead I was able to make it work. Thanks a lot for your help!!!

closed 2021-06-19 by nocoinman