gazebosim/gazebo-classic #3028: Bug: wind_demo.world does not demonstrate wind.

acxz, 2021-06-19:
I am attempting to use the `<wind>  <linear_velocity>` feature in gazebo. In lieu of that I tried running `wind_demo.world` but saw no effect of the wind.

Command:
`gazebo /usr/share/gazebo-11/worlds/wind_demo.world`

Expected result: See the spheres moving in the direction of the wind.

Actual result: No motion occurs throughout the entirety of the simulation.

Gazebo version: 11.6.0

chapulina, 2021-06-30:
I see this error when running `--verbose`:

`[Err] [WindPlugin.cc:212] Please set <force_approximation_scaling_factor> to a value greater than 0`

It looks like that element was made mandatory on Gazebo 10 but the example world was not updated.

acxz, 2021-06-30:
Ahh `--verbose`, totally forgot about that.

Yep, adding `force_approximation_scaling_factor` lets the world behave as expected. I'll make a PR to close this issue if that's okay.