gazebosim/gazebo-classic #3043: bug: SimpleTrackedVehiclePlugin broken ?

charles-fox, 2021-07-13:
I just tried a fresh install of Ubuntu 20.04 then gazebo install (no ROS), via curl -sSL http://get.gazebosim.org | sh .

It looks like the SimpleTrackedVehiclePlugin is broken in the current release, when I run the demo,
/usr/share/gazebo-11/worlds/tracked_vehicle_simple.world

the vehicle doesn't move any more.   I can see the keyboard plugin is working, and cmd_vel_twist gz messages are working, but the plugin isn't making the vehicle move any more.

Does anyone else see this with this setup?  Any fixes known?

lprobsth, 2022-07-26:
I have the same problem - SimpleTrackedVehiclePlugin also publishes the track speeds as expected...

lprobsth, 2022-07-27:
I tested two scenarios:
1. gazebo through roslaunch: works (previously my SDF referenced the wrong robot frame so SimpleTrackedVehiclePlugin was not able to perform the movement - don't know why no error was visible...)
2. launching gazebo in standalone mode with the world file provided with the standard gazebo installation (the world file you tested)

Initially the plugin didn't work. There was no output in the command line. I then started gazebo with verbose output (```gazebo --verbose /usr/share/gazebo-11/worlds/tracked_vehicle_simple.world```).

```
Gazebo multi-robot simulator, version 11.11.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.11.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.36
[Msg] Loading world file [/usr/share/gazebo-11/worlds/tracked_vehicle_simple.world]
[Err] [Plugin.hh:212] Failed to load plugin /usr/lib/x86_64-linux-gnu/gazebo-11/plugins/libSimpleTrackedVehiclePlugin.so: libTrackedVehiclePlugin.so: cannot open shared object file: No such file or directory
[Msg] keyboard_control Plugin <publish_as_twist> set to 1
[Msg] keyboard_control Plugin <cmd_vel_topic> set to ~/simple_tracked/cmd_vel_twist
[Msg] keyboard_control Plugin missing <max_linear_vel>, defaults to 1
[Msg] keyboard_control Plugin missing <min_linear_vel>, defaults to -1
[Msg] keyboard_control Plugin missing <linear_increment>, defaults to 0.5
[Msg] keyboard_control Plugin missing <max_angular_vel>, defaults to 1
[...]
```

I then looked at the LD_LIBRARY_PATH environment variable:
```LD_LIBRARY_PATH=/home/user/test_ws/devel/lib:/home/user/base_ws/devel/lib:/opt/ros/noetic/lib```

-> the shared library path of gazebo is not in the environment variable

when I source the gazebo setup file with ```source /usr/share/gazebo-11/setup.bash``` the robot moves as expected.

It seems that the plugin works - so this issue probably can be closed.

If you still experience this issue try to enabled the debug output in the roslaunch file or with the "--verbose" option and post the output.

Vineet-Pandey, 2022-12-12:
Was this bug fixed? Because I am seeing the same behaviour in gazebo 9.19. * I tried sourcing the setup file as @lprobsth suggested but that didn't fix the problem*

lprobsth, 2022-12-12:
Please post the output of the test world with the "--verbose" output. There could be many reasons for the plugin to fail.

Vineet-Pandey, 2022-12-12:
I am receiving the following output in the verbose mode:
```
[Msg] simple_tracked_vehicle Plugin missing <robot_namespace>, defaults to robot_setter
[Msg] simple_tracked_vehicle Plugin missing <steering_efficiency>, defaults to 0.5
[Msg] simple_tracked_vehicle Plugin <tracks_separation> set to 1.914
[Msg] simple_tracked_vehicle Plugin <max_linear_speed> set to 15
[Msg] simple_tracked_vehicle Plugin <max_angular_speed> set to 2
[Msg] simple_tracked_vehicle Plugin <track_mu> set to 2
[Msg] simple_tracked_vehicle Plugin <track_mu2> set to 2
[Msg] SimpleTrackedVehiclePlugin: Successfully added robot body link ps0_base_link
[Msg] SimpleTrackedVehiclePlugin: Successfully added left track link ps0_left_track_link
[Msg] SimpleTrackedVehiclePlugin: Successfully added right track link ps0_right_track_link
[Msg] simple_tracked_vehicle Plugin missing <collide_without_contact_bitmask>, defaults to 1

```

It appears to have been launched correctly. Any insights?