gazebosim/gazebo-classic #3045: Gazebo requires X11 to run

icefoxen, 2021-07-17:
When I try to run `gzclient` on a Linux computer using Sway or another Wayland window manager, it crashes with the following message: `terminate called after throwing an instance of 'gazebo::common::Exception'.  “gzclient” terminated by signal SIGABRT (Abort)`.  This appears to be the same as this bug: https://bugzilla.redhat.com/show_bug.cgi?id=1816487 and has the same resolution: running `XDG_SESSION_TYPE=x11 gazebo` works.  That old bug report is for Gazebo 9 and I am seeing this is still a problem using gazebo-11.1.0 on Debian 11.

Sabrin0, 2023-06-07:
Hi @icefoxen, did you solve the issue? Because I'm using Gazebo 11 and I got the same error 😢

eliasffyksen, 2023-06-07:
If you just want to run it, have you tried running it with `XDG_SESSION_TYPE=x11 gazebo` as the fist comment suggests. I had this issue and this solved it. iirc the problem had to do with gazebo not thinking that it can run in a wayland session (while it's actually fine), adding the `XDG_SESSION_TYPE=x11` makes it "think" that it is in a X11 session. This solution worked find for me at least