gazebosim/gazebo-classic #3046: Gazebo 11.7.0 does not compile against qwt 6.2

traversaro, 2021-07-19:
Build error:
~~~
2021-07-19T22:13:26.6397288Z ../gazebo/gui/plot/IncrementalPlot.cc: In member function 'virtual void gazebo::gui::PlotMagnifier::rescale(double)':
2021-07-19T22:13:26.6398542Z ../gazebo/gui/plot/IncrementalPlot.cc:83:35: error: variable 'const QwtScaleMap xScaleMap' has initializer but incomplete type
2021-07-19T22:13:26.6399901Z    83 |                 const QwtScaleMap xScaleMap = plt->canvasMap(QwtPlot::xBottom);
2021-07-19T22:13:26.6400453Z       |                                   ^~~~~~~~~
2021-07-19T22:13:26.6401166Z ../gazebo/gui/plot/IncrementalPlot.cc:83:78: error: invalid use of incomplete type 'class QwtScaleMap'
2021-07-19T22:13:26.6401973Z    83 |                 const QwtScaleMap xScaleMap = plt->canvasMap(QwtPlot::xBottom);
2021-07-19T22:13:26.6402536Z       |                                                                              ^
2021-07-19T22:13:26.6403029Z In file included from ../gazebo/gui/plot/qwt_gazebo.h:30,
2021-07-19T22:13:26.6403534Z                  from ../gazebo/gui/plot/PlotTracker.hh:23,
2021-07-19T22:13:26.6403990Z                  from ../gazebo/gui/plot/IncrementalPlot.cc:26:
2021-07-19T22:13:26.6404691Z $PREFIX/include/qwt_painter.h:19:7: note: forward declaration of 'class QwtScaleMap'
2021-07-19T22:13:26.6405153Z    19 | class QwtScaleMap;
2021-07-19T22:13:26.6405508Z       |       ^~~~~~~~~~~
2021-07-19T22:13:26.6406220Z ../gazebo/gui/plot/IncrementalPlot.cc:84:35: error: variable 'const QwtScaleMap yScaleMap' has initializer but incomplete type
2021-07-19T22:13:26.6407049Z    84 |                 const QwtScaleMap yScaleMap = plt->canvasMap(QwtPlot::yLeft);
2021-07-19T22:13:26.6407581Z       |                                   ^~~~~~~~~
2021-07-19T22:13:26.6408280Z ../gazebo/gui/plot/IncrementalPlot.cc:84:76: error: invalid use of incomplete type 'class QwtScaleMap'
2021-07-19T22:13:26.6409084Z    84 |                 const QwtScaleMap yScaleMap = plt->canvasMap(QwtPlot::yLeft);
2021-07-19T22:13:26.6409650Z       |                                                                            ^
2021-07-19T22:13:26.6410120Z In file included from ../gazebo/gui/plot/qwt_gazebo.h:30,
2021-07-19T22:13:26.6410575Z                  from ../gazebo/gui/plot/PlotTracker.hh:23,
2021-07-19T22:13:26.6411147Z                  from ../gazebo/gui/plot/IncrementalPlot.cc:26:
2021-07-19T22:13:26.6411857Z $PREFIX/include/qwt_painter.h:19:7: note: forward declaration of 'class QwtScaleMap'
2021-07-19T22:13:26.6412311Z    19 | class QwtScaleMap;
2021-07-19T22:13:26.6412666Z       |       ^~~~~~~~~~~
2021-07-19T22:13:26.6413400Z ../gazebo/gui/plot/IncrementalPlot.cc: In member function 'void gazebo::gui::IncrementalPlot::Update()':
2021-07-19T22:13:26.6414286Z ../gazebo/gui/plot/IncrementalPlot.cc:371:21: error: variable 'const QwtScaleMap xScaleMap' has initializer but incomplete type
2021-07-19T22:13:26.6415093Z   371 |   const QwtScaleMap xScaleMap = this->canvasMap(QwtPlot::xBottom);
2021-07-19T22:13:26.6415572Z       |                     ^~~~~~~~~
2021-07-19T22:13:26.6416299Z ../gazebo/gui/plot/IncrementalPlot.cc:371:65: error: invalid use of incomplete type 'class QwtScaleMap'
2021-07-19T22:13:26.6417068Z   371 |   const QwtScaleMap xScaleMap = this->canvasMap(QwtPlot::xBottom);
2021-07-19T22:13:26.6417593Z       |                                                                 ^
2021-07-19T22:13:26.6418060Z In file included from ../gazebo/gui/plot/qwt_gazebo.h:30,
2021-07-19T22:13:26.6418505Z                  from ../gazebo/gui/plot/PlotTracker.hh:23,
2021-07-19T22:13:26.6418974Z                  from ../gazebo/gui/plot/IncrementalPlot.cc:26:
2021-07-19T22:13:26.6419666Z $PREFIX/include/qwt_painter.h:19:7: note: forward declaration of 'class QwtScaleMap'
2021-07-19T22:13:26.6420120Z    19 | class QwtScaleMap;
2021-07-19T22:13:26.6420473Z       |       ^~~~~~~~~~~
~~~

First detected in conda-forge (see https://github.com/conda-forge/gazebo-feedstock/pull/85). We are going to ping to qwt 6.1 there, but it is possible that this issue will soon hit Homebrew, if qwt is bumped there, fyi @scpeters .

scpeters, 2021-07-20:
I believe it will be fixed by https://github.com/scpeters/gazebo/commit/70f013777b9d93f1e4c8d7a6b1cbcf23ff9692cc in https://github.com/osrf/gazebo/pull/3047

scpeters, 2021-07-22:
fixed by #3047

closed 2021-07-22 by scpeters