gazebosim/gazebo-classic #3055: Issues when building Gazebo

mboisson, 2021-08-04:
Hi,
I have many issues when trying to build Gazebo 11, and I am hoping someone can give me pointers. @traversaro suggested I post here. 

I have the following dependencies installed: 
```
 18 builddependencies = [
 19     ('CMake', '3.16.4'),
 21     ('protobuf', '3.12.3'),
 22     ('tbb', '2020.2'),
 23     ('Simbody', '3.7'),
 25     ('GTS', '20121130'),
 26     ('Bullet', '2.89'),
 27     ('libccd', '2.1'),
 28     ('Qt5', '5.12.8'),
 29     ('OGRE', '1.12.12'),
 30     ('ignition', 'citadel'),
 31     ('Boost', '1.72.0'),
 32     ('Qwt', '6.1.4'),
 33     ('HDF5', '1.10.6'),
 34     ('GDAL', '3.0.4'),
 36 ]
``` 

The complete build log is here: 
https://gist.github.com/mboisson/5a2375c8fbecae27b430e6215dbc4b21

The first issue seems that it is not finding TBB properly: 
```
 576 CMake Error at cmake/GazeboUtils.cmake:99 (add_library):
 577   Target "libgazebo_client" links to target "TBB::tbb" but the target was not
 578   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
 579   an ALIAS target is missing?
 580 Call Stack (most recent call first):
 581   gazebo/CMakeLists.txt:130 (gz_add_library)
``` 
so... it is finding `TBB::tbb`, and links against it, but it's not finding it ? 

The other issue is that it hardcodes `/usr/lib` somewhere: 
```
2582 CMake Warning at cmake/GazeboTestUtils.cmake:42 (add_executable):
2583   Cannot generate a safe runtime search path for target
2584   UNIT_AudioDecoder_TEST because files in some directories may conflict with
2585   libraries in implicit directories:
2586
2587     runtime library [libdl.so.2] in /cvmfs/soft.computecanada.ca/gentoo/2020/usr/lib may be hidden by files in:
2588       /usr/lib
2589
2590   Some of these libraries may not be found correctly.
``` 

It should not be using `/usr/lib` at all. Those are not the proper libraries to use on our clusters. 

Any hint at how I may fix that ?

mboisson, 2021-08-04:
So, for the `/usr/lib` part, I managed to fix it by specifying explicitly: 
```
-Dtinyxml_LIBDIR=$EBROOTGENTOO/lib -Dtinyxml_LIBRARY_DIRS=$EBROOTGENTOO/lib -Dtinyxml_PREFIX=$EBROOTGENTOO
``` 

For some reason, CMake found `tinyxml` in the right location (in `$EBROOTGENTOO`), but failed to set variables accordingly.  You may want to have a look at how `tinyxml` is detected and variables set. 

For the `tbb` errors, this is in fact due to a bug in the way TBB's cmake support is done, which defaults to having other applications that configure TBB overwrite its own CMake files... and our build was corrupted by a later install.

traversaro, 2021-08-04:
For the `tinyxml`/`/usr/lib` part, can you copy&paste the `CMakeCache.txt` of the build when this was happening? Having a look on how tinyxml is handled without any clue on how things are happening in your setup may not be trivial.

mboisson, 2021-08-04:
Sure. Here it is: 
https://gist.github.com/mboisson/d2e5e4189354d4ca82a1170be455170f

See that it found: 
```
pkgcfg_lib_tinyxml_tinyxml:FILEPATH=/cvmfs/soft.computecanada.ca/gentoo/2020/usr/lib/libtinyxml.so
``` 

but then it set: 
```
tinyxml_CFLAGS:INTERNAL=-DTIXML_USE_STL=YES;-I/usr/include
tinyxml_CFLAGS_I:INTERNAL=
tinyxml_CFLAGS_OTHER:INTERNAL=-DTIXML_USE_STL=YES
tinyxml_FOUND:INTERNAL=1
tinyxml_INCLUDEDIR:INTERNAL=/usr/include
tinyxml_INCLUDE_DIRS:INTERNAL=/usr/include
tinyxml_LDFLAGS:INTERNAL=-L/usr/lib;-ltinyxml
tinyxml_LDFLAGS_OTHER:INTERNAL=
tinyxml_LIBDIR:INTERNAL=/usr/lib
tinyxml_LIBRARIES:INTERNAL=tinyxml
tinyxml_LIBRARY_DIRS:INTERNAL=/usr/lib
tinyxml_LIBS:INTERNAL=
tinyxml_LIBS_L:INTERNAL=
tinyxml_LIBS_OTHER:INTERNAL=
tinyxml_LIBS_PATHS:INTERNAL=
tinyxml_MODULE_NAME:INTERNAL=tinyxml
tinyxml_PREFIX:INTERNAL=/usr
tinyxml_STATIC_CFLAGS:INTERNAL=-DTIXML_USE_STL=YES;-I/usr/include
tinyxml_STATIC_CFLAGS_I:INTERNAL=
tinyxml_STATIC_CFLAGS_OTHER:INTERNAL=-DTIXML_USE_STL=YES
tinyxml_STATIC_INCLUDE_DIRS:INTERNAL=/usr/include
tinyxml_STATIC_LDFLAGS:INTERNAL=-L/usr/lib;-ltinyxml
tinyxml_STATIC_LDFLAGS_OTHER:INTERNAL=
tinyxml_STATIC_LIBDIR:INTERNAL=
tinyxml_STATIC_LIBRARIES:INTERNAL=tinyxml
tinyxml_STATIC_LIBRARY_DIRS:INTERNAL=/usr/lib
tinyxml_STATIC_LIBS:INTERNAL=
tinyxml_STATIC_LIBS_L:INTERNAL=
tinyxml_STATIC_LIBS_OTHER:INTERNAL=
tinyxml_STATIC_LIBS_PATHS:INTERNAL=
tinyxml_VERSION:INTERNAL=2.6.2
tinyxml_tinyxml_INCLUDEDIR:INTERNAL=
tinyxml_tinyxml_LIBDIR:INTERNAL=
tinyxml_tinyxml_PREFIX:INTERNAL=
tinyxml_tinyxml_VERSION:INTERNAL=
```

traversaro, 2021-08-04:
Are you sure you don't have a malformed tinyxml.pc in your install of tinyxml?

mboisson, 2021-08-04:
We don't have a `tinyxml.pc` file. CMake apparently finds the library not through pkgconfig.

mboisson, 2021-08-04:
Indeed. Apparently, the `tinyxml.pc` file installed in our Gentoo Prefix installation is bonker... wonder why we did not encounter that before 🤔

mboisson, 2021-08-04:
That fixed it. Thanks for rubber ducking. I will close this now.

closed 2021-08-04 by mboisson