gazebosim/gazebo-classic #3068: Version conflict in arm64 binaries

pgorczak, 2021-08-18:
I just failed trying to install on arm64 and the package versions don't line up. Snippet from [Packages](http://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/focal/main/binary-arm64/Packages)

```
Package: gazebo11
Version: 11.7.0-1~focal
Architecture: arm64
Maintainer: Nate Koenig <nate@osrfoundation.org>
Installed-Size: 7497
Depends: [...] gazebo11-common (= 11.7.0-1~focal)
[...]

Package: gazebo11-common
Source: gazebo11
Version: 11.8.0-1~focal
[...]
```

j-rivero, 2021-08-18:
Hello Phillip:

There is a new release ongoing for gazebo11, 11.8.0. The builder job for arm64 is still in progress, should finish in the next hours: https://build.osrfoundation.org/job/gazebo11-debbuilder/112 

To avoid this problem in the future I'm going to merge https://github.com/ignition-release/gazebo11-release/pull/6/files to relax the restriction on the `-common-` package.

Thanks.

pgorczak, 2021-08-18:
Hi Jose, thanks for the explanation. I'll be watching the Jenkins build for 11.8.0 and close this when its done, just in case anyone stumbles upon the same issue.

closed 2021-08-18 by pgorczak