gazebosim/gazebo-classic #3115: Huge incoherences when reading joint/link positions and velocities (maybe Gazebo state handling error)

AndPatr, 2021-10-06:
I have found a potentially very serious issue when reading the states of my robot.
To exclude potential errors in my plugin code, I have also used the plotting utility of the GUI.
You can see the details at [https://answers.gazebosim.org/question/27430/huge-incoherences-when-reading-jointlink-positions-and-velocities-probably-gazebo-internal-state-handling-error/](url)

scpeters, 2021-10-12:
I'll be more specific with my questions:

1. Are you using SDFormat or URDF for your model?
2. What joints are in your model? Are you using revolute joints for all parts of the closed kinematic loop?
3. What gazebo API's are you using for joint control?

AndPatr, 2021-10-13:
Well,
I didn't use SetPosition since apparently does not work on closed kinematic chains (at the time I tested it and didn't work).
In my plugin I make use of:
- this->hinge_name->Position(0) and this->hinge_name->GetVelocity(0) for reading joint positions and velocities
- this->left_wheel_state=physics::LinkState(this->left_wheel) and the Pose() and Velocity() methods to get the necessary link pose and velocity
- I use Joint::SetForce for applying control inputs (which are torques in my case)

AndPatr, 2021-10-13:
If it helps, I can share the plugin, the world file and the model.sdf file.