gazebosim/gazebo-classic #3130: run gazebo error

chongjeelee, 2021-11-09:
when i input 'gazebo' in terminal,it ocurred this.
[image]
who can i solve this problem?thanks.

traversaro, 2021-11-09:
Hi @chongjeelee, you can probably get more information (and provide it in the issue) by running `gazebo --verbose` .

chongjeelee, 2021-11-09:
> Hi @chongjeelee, you can probably get more information (and provide it in the issue) by running `gazebo --verbose` .

[image]
ok,i can start gazebo,but i find a new problem...
[image]
i can not get the topics from gazebo in terminal (rostopic list)

traversaro, 2021-11-09:
Even in this case, if you could run `gazebo --verbose` you will get more information that will be helpful for you and for anyone else to debug the problem. However, if this is not a specific bug of Gazebo but rather a specific problem of your setup, you can consider posting in https://answers.gazebosim.org/questions/ instead.

nocoinman, 2021-11-09:
@chongjeelee 

Running `gazebo` directly from the terminal won't work with ROS. You'll need to launch gazebo using the ROS Interface, ie 
[`gazebo_ros_pkgs`](https://github.com/ros-simulation/gazebo_ros_pkgs).

To make it more maintainable, you can write a launch file dedicated to launching gazebo with your desired flags.
Here is a sample template

```xml
<launch>
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="world_name" value="$(find your-pkg-with-worlds)/worlds/your-world-file-name.world"/>
        <arg name="paused" value="true"/>   <!-- Starts gazebo with physics paused -->
        <arg name="use_sim_time" value="true"/>
        <arg name="gui" value="true"/>
        <arg name="headless" value="false"/>
        <arg name="debug" value="false"/>
        <arg name="verbose" value="true"/>
    </include>
</launch>
```

The arguments are all self explanatory. If you need further info you can consult `gazebo_ros_pkgs` documentation.

chongjeelee, 2021-11-09:
> @chongjeelee
> 
> Running `gazebo` directly from the terminal won't work with ROS. You'll need to launch gazebo using the ROS Interface, ie [`gazebo_ros_pkgs`](https://github.com/ros-simulation/gazebo_ros_pkgs).
> 
> To make it more maintainable, you can write a launch file dedicated to launching gazebo with your desired flags. Here is a sample template
> 
> ```
> <launch>
>     <include file="$(find gazebo_ros)/launch/empty_world.launch">
>         <arg name="world_name" value="$(find your-pkg-with-worlds)/worlds/your-world-file-name.world"/>
>         <arg name="paused" value="true"/>   
>         <arg name="use_sim_time" value="true"/>
>         <arg name="gui" value="true"/>
>         <arg name="headless" value="false"/>
>         <arg name="debug" value="false"/>
>         <arg name="verbose" value="true"/>
>     </include>
> </launch>
> ```
> 
> The arguments are all self explanatory. If you need further info you can consult `gazebo_ros_pkgs` documentation.

good! it does work,thank you so much!

closed 2021-11-09 by chongjeelee