gazebosim/gazebo-classic #3133: Issue installing Gazebo 11 on Ubuntu 20.04

krishna-py, 2021-11-19:
Summary:
I am trying to install gazebo 11 using the steps highlighted in [here](http://gazebosim.org/tutorials?tut=install_ubuntu&cat=install)
I am getting this error when the script tries to install libgazebo11-dev. If I ignore this package, I am able to install gazebo11 and libgazebo11.

My OS is a clean Ubuntu 20.04. I only installed build-essentials before this.

$> curl -sSL http://get.gazebosim.org | sh

In Debian this script will setup the osrfoundation
repository to install the latest package available

+ sudo -E sh -c apt-key adv --keyserver keyserver.ubuntu.com --recv-keys D2486D2DD83DB69272AFE98867170598AF249743
Executing: /tmp/apt-key-gpghome.kbFbcfs6gy/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys D2486D2DD83DB69272AFE98867170598AF249743
gpg: key 67170598AF249743: "OSRF Repository (OSRF Repository GPG key) <osrfbuild@osrfoundation.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
+ sudo -E sh -c mkdir -p /etc/apt/sources.list.d
+ sudo -E sh -c echo deb http://packages.osrfoundation.org/gazebo/ubuntu\-stable focal main > /etc/apt/sources.list.d/gazebo-stable.list
+ sudo -E sh -c sleep 3; apt-get update; apt-get install -y -q libgazebo11-dev gazebo11
Hit:1 http://dl.google.com/linux/chrome/deb stable InRelease
Hit:2 http://in.archive.ubuntu.com/ubuntu focal InRelease                                                                                 
Get:3 http://packages.osrfoundation.org/gazebo/ubuntu-stable focal InRelease [4,249 B]                                                    
Hit:4 http://security.ubuntu.com/ubuntu focal-security InRelease
Get:5 http://packages.osrfoundation.org/gazebo/ubuntu-stable focal/main i386 Packages [4,913 B]
Get:6 http://packages.osrfoundation.org/gazebo/ubuntu-stable focal/main amd64 Packages [67.6 kB]
Fetched 76.7 kB in 2s (39.5 kB/s)   
Reading package lists... Done
Reading package lists...
Building dependency tree...
Reading state information...
gazebo11 is already the newest version (11.9.0-1~focal).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgazebo11-dev : Depends: qtbase5-dev but it is not going to be installed
                   Depends: libprotoc-dev but it is not going to be installed
                   Depends: libprotobuf-dev but it is not going to be installed
                   Depends: libboost-all-dev but it is not going to be installed
                   Depends: freeglut3-dev but it is not going to be installed
                   Depends: libgts-dev but it is not going to be installed
                   Depends: libgdal-dev but it is not going to be installed
                   Depends: libignition-transport8-dev but it is not going to be installed
                   Depends: libignition-msgs5-dev but it is not going to be installed
                   Depends: libignition-fuel-tools4-dev but it is not going to be installed
                   Depends: libignition-common3-dev but it is not going to be installed
                   Depends: libignition-common3-graphics-dev but it is not going to be installed
                   Depends: libignition-common3-profiler-dev but it is not going to be installed
                   Depends: libsimbody-dev (>= 3.0.0) but it is not going to be installed
                   Depends: libsimbody-dev (< 4.0.0) but it is not going to be installed
                   Depends: libdart6-collision-bullet-dev (< 6.10.0) but it is not installable or
                            libdart-collision-bullet-dev (< 6.10.0) but it is not going to be installed
                   Depends: libdart6-collision-ode-dev (< 6.10.0) but it is not installable or
                            libdart-collision-ode-dev (< 6.10.0) but it is not going to be installed
                   Depends: libdart6-utils-urdf-dev (< 6.10.0) but it is not installable or
                            libdart-utils-urdf-dev (< 6.10.0) but it is not going to be installed
                   Depends: libdart6-external-odelcpsolver-dev (< 6.10.0) but it is not installable or
                            libdart-external-odelcpsolver-dev (< 6.10.0) but it is not going to be installed
                   Depends: libdart6-external-ikfast-dev (< 6.10.0) but it is not installable or
                            libdart-external-ikfast-dev (< 6.10.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

----------------------------
$> lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

----------------------------
When I exclude libgazebo11-dev from the sh file, i am able to install. I need this package for development.
$> dpkg -l | grep gazebo
ii  gazebo11                                   11.9.0-1~focal                        amd64        Open Source Robotics Simulator
ii  gazebo11-common                            11.9.0-1~focal                        all          Open Source Robotics Simulator - Shared files
ii  gazebo11-plugin-base                       11.9.0-1~focal                        amd64        Open Source Robotics Simulator - base plug-ins
ii  libgazebo11:amd64                          11.9.0-1~focal                        amd64        Open Source Robotics Simulator - shared library

chapulina, 2021-12-28:
I can't reproduce the issue on a 20.04 machine.

One way to track down the culprit is to install each of the "_but it is not going to be installed_" dependencies and slowly narrow down where the issue is.

krishna-py, 2021-12-28:
I found the reason for it. My original Ubuntu image ISO was 18.04 using which I did a minimal install. Then I upgraded to 20.04 using the OS upgrade option. I think this is causing the issue as somewhere the install file was identifying the OS version wrongly, even though lsb_release -a is still saying OS is Ubuntu 20.04.

Closing the issue, as I installed from a 20.04 image directly and it is working as expected.

closed 2021-12-28 by krishna-py