gazebosim/gazebo-classic #3141: gazebo/msgs/poses_stamped.pb.h: No such file or directory on cmake 3.22 when BUILD_TESTING is enabled

alexfanqi, 2021-12-02:
Hi there,

I am testing gazebo-11.7.0 on Gentoo. With cmake 3.22 and BUILD_TESTING=TRUE, it fails with
```
In file included from /var/tmp/portage/sci-electronics/gazebo-11.7.0-r1/work/gazebo-11.7.0/gazebo/common/Plugin.hh:41,
                 from /var/tmp/portage/sci-electronics/gazebo-11.7.0-r1/work/gazebo-11.7.0/gazebo/common/Plugin_TEST.cc:18:
/var/tmp/portage/sci-electronics/gazebo-11.7.0-r1/work/gazebo-11.7.0/gazebo/physics/PhysicsTypes.hh:26:10: fatal error: gazebo/msgs/poses_stamped.pb.h: No such file or directory
   26 | #include "gazebo/msgs/poses_stamped.pb.h"
```
seems like gazebo/common is compiled before msgs/*.pb.h is generated.

It builds successfully on cmake 3.21.4, or with BUILD_TESTING=FALSE though.

Please find full compilation log and build environment information in https://bugs.gentoo.org/827964

Terence-I, 2023-03-30:
getting the same error if anyone can help

traversaro, 2023-03-30:
I also experienced  a similar error. A workaround is to manually run make gazebo_msgs or ninja gazebo_msgs before actually compiling the test.