gazebosim/gazebo-classic #3144: Record utility fails to capture velocity, acceleration, wrench

shonigmann, 2021-12-06:
I'm trying to record logs of simulations to be able to automatically extract model dynamics for monte-carlo type simulations. The log utility seems perfect for this case, as it can be started with gzserver and run for a given number of iterations... however, I cannot for the life of me replicate the results demonstrated in the Gazebo [Log Filtering](http://www.gazebosim.org/tutorials?tut=log_filtering&cat=tools_utilities) and [Logging Playback ](https://gazebosim.org/tutorials?cat=tools_utilities&tut=logging_playback) tutorials, which state and show that you can record pose, _velocity, acceleration, and wrench_ data for links in the simulation. Inspecting the generated log files, I notice that _only pose_ data is captured. Filtering explicitly for the other terms shows that they are 0 for each log update. 

OS: Ubuntu 20.04
Gazebo version: 11.9

Steps to reproduce:
```
# remove existing logs
rm -rf ~/.gazebo/log/*  

# run cart_demo world; note the debug info shows non-zero velocities
gzserver -r worlds/cart_demo.world --verbose  

#kill after a few seconds of data have been collected

# echo log file, filtering for velocity
gz log -e -f ~/.gazebo/log/*/gzserver/state.log --filter */*.velocity -z 10
```

Expected output:
non-zero velocity logged for each link

Actual output:
```
<chunk encoding='txt'><![CDATA[
<sdf version='1.7'>
<state world_name='default'>
<sim_time>1 224000000</sim_time>
<real_time>1 279422157</real_time>
<wall_time>1638826147 920945581</wall_time>
<iterations>1224</iterations>
<model name='ground_plane'>
<link name='link'>
<velocity>0.000000 0.000000 0.000000 0.000000 -0.000000 0.000000</velocity>
</link>
</model>
<model name='test_model'>
<link name='bar_1'>
<velocity>0.000000 0.000000 0.000000 0.000000 -0.000000 0.000000</velocity>
</link>
<link name='bar_2'>
<velocity>0.000000 0.000000 0.000000 0.000000 -0.000000 0.000000</velocity>
</link>
<link name='bar_3'>
<velocity>0.000000 0.000000 0.000000 0.000000 -0.000000 0.000000</velocity>
</link>
<link name='cylinder_link'>
<velocity>0.000000 0.000000 0.000000 0.000000 -0.000000 0.000000</velocity>
</link>
<link name='link'>
<velocity>0.000000 0.000000 0.000000 0.000000 -0.000000 0.000000</velocity>
</link>
<link name='sphere_link'>
<velocity>0.000000 0.000000 0.000000 0.000000 -0.000000 0.000000</velocity>
</link>
<link name='wheel_1'>
<velocity>0.000000 0.000000 0.000000 0.000000 -0.000000 0.000000</velocity>
</link>
...
```

HKenta, 2022-09-05:
Hi,

I use gazebo9, but I'm also faced this trouble.
Did you solve this, or get any information?

shonigmann, 2022-09-06:
for my purposes, I ended up writing a plugin that wrote data to a .csv file using the [Gazebo Physics Link API](http://osrf-distributions.s3.amazonaws.com/gazebo/api/9.0.0/classgazebo_1_1physics_1_1Link.html)  

You can find it here, if it's helpful: https://github.com/shonigmann/gazebo_plugins/blob/main/src/gazebo_log_state.cpp

HKenta, 2022-09-07:
Thank you for replying. It seems helpful, I'll check it.