gazebosim/gazebo-classic #3159: Cannot change the heightmap while running

woensug-choi, 2022-01-14:
## Objective that can't be achieved
Remove the old heightmap and insert a new heightmap in the scene while running. 

## To reproduce
- From gazebo client
Open any world with heightmap (maybe [this tutorial](http://gazebosim.org/tutorials/?tut=dem))
Nothing in the scene changes even with 'remove model' on the gazebo client panel.

- From the gazebo plugin
The function I am using from the gazebo plugin side are
```
world->InsertModelFile('ModelName')
world->RemoveModel('ModelName')
```
which should be a generic method but nothing changes on the screen.

## Background
This feature is essential, when we are working with maritime vehicles operating in a large area (10~50km range). How the [dave project](https://github.com/Field-Robotics-Lab/dave) implemented is to dissect the large GeoTiff bathymetry into tiles and auto spawn and remove according to the vehicle location. We've been doing that by generating the mesh (obj) based model for each tile but wanted to migrate to using the heightmap rendering/physics (which looks much appealing with the Ogre engine at the backend).

@scpeters @WilliamLewww

scpeters, 2022-01-20:
In my testing, it appears that the heightmap collisions are deleted from physics, but the visuals are not deleted from the rendering system.

it seems that `gazebo::rendering::Heightmap` [does not inherit from `gazebo::rendering::Visual`](https://github.com/osrf/gazebo/blob/gazebo11_11.10.0/gazebo/rendering/Heightmap.hh#L62) and requires [special handling with a dummy Visual when being added to a Scene](https://github.com/osrf/gazebo/blob/gazebo11_11.10.0/gazebo/rendering/Scene.cc#L2843-L2848).

I think the logic for responding to `"entity_delete"` requests in [Scene::ProcessRequestMsg](https://github.com/osrf/gazebo/blob/gazebo11_11.10.0/gazebo/rendering/Scene.cc#L2564-L2583) needs to be updated to explicitly remove the heightmap from the scene. It may be as simple as copying [two lines from `Scene::Clear()](https://github.com/osrf/gazebo/blob/gazebo11_11.10.0/gazebo/rendering/Scene.cc#L301-L302).

scpeters, 2022-02-11:
this should be fixed by #3171 and will be available with the next gazebo11 release

closed 2022-02-11 by scpeters

scpeters, 2022-02-14:
@woensug-choi this has been fixed but not yet released. Is it urgent to make a patch release with the fix for this bug?

woensug-choi, 2022-02-14:
@scpeters Not at all :) Thank you for the quick fixes!