gazebosim/gazebo-classic #3170: Unresolved externals when linking with boost in Windows CI

scpeters, 2022-02-08:
Windows CI jobs are failing:

[image] https://build.osrfoundation.org/job/gazebo-ci-pr_any-windows7-amd64/3456/

~~~
   Creating library gazebo_physics.lib and object gazebo_physics.exp
ModelState.cc.obj : error LNK2019: unresolved external symbol "public: bool __cdecl boost::re_detail_107500::perl_matcher<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class std::allocator<struct boost::sub_match<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > > > >,struct boost::regex_traits<char,class boost::w32_regex_traits<char> > >::match(void)" (?match@?$perl_matcher@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@V?$allocator@U?$sub_match@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@@boost@@@2@U?$regex_traits@DV?$w32_regex_traits@D@boost@@@boost@@@re_detail_107500@boost@@QEAA_NXZ) referenced in function "bool __cdecl boost::regex_match<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class std::allocator<struct boost::sub_match<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > > > >,char,struct boost::regex_traits<char,class boost::w32_regex_traits<char> > >(class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class boost::match_results<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class std::allocator<struct boost::sub_match<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > > > > > &,class boost::basic_regex<char,struct boost::regex_traits<char,class boost::w32_regex_traits<char> > > const &,enum boost::regex_constants::_match_flags)" (??$regex_match@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@V?$allocator@U?$sub_match@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@@boost@@@2@DU?$regex_traits@DV?$w32_regex_traits@D@boost@@@boost@@@boost@@YA_NV?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@0AEAV?$match_results@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@V?$allocator@U?$sub_match@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@@boost@@@2@@0@AEBV?$basic_regex@DU?$regex_traits@DV?$w32_regex_traits@D@boost@@@boost@@@0@W4_match_flags@regex_constants@0@@Z)
WorldState.cc.obj : error LNK2001: unresolved external symbol "public: bool __cdecl boost::re_detail_107500::perl_matcher<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class std::allocator<struct boost::sub_match<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > > > >,struct boost::regex_traits<char,class boost::w32_regex_traits<char> > >::match(void)" (?match@?$perl_matcher@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@V?$allocator@U?$sub_match@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@@boost@@@2@U?$regex_traits@DV?$w32_regex_traits@D@boost@@@boost@@@re_detail_107500@boost@@QEAA_NXZ)
ModelState.cc.obj : error LNK2019: unresolved external symbol "private: void __cdecl boost::re_detail_107500::perl_matcher<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class std::allocator<struct boost::sub_match<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > > > >,struct boost::regex_traits<char,class boost::w32_regex_traits<char> > >::construct_init(class boost::basic_regex<char,struct boost::regex_traits<char,class boost::w32_regex_traits<char> > > const &,enum boost::regex_constants::_match_flags)" (?construct_init@?$perl_matcher@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@V?$allocator@U?$sub_match@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@@boost@@@2@U?$regex_traits@DV?$w32_regex_traits@D@boost@@@boost@@@re_detail_107500@boost@@AEAAXAEBV?$basic_regex@DU?$regex_traits@DV?$w32_regex_traits@D@boost@@@boost@@@3@W4_match_flags@regex_constants@3@@Z) referenced in function "bool __cdecl boost::regex_match<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class std::allocator<struct boost::sub_match<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > > > >,char,struct boost::regex_traits<char,class boost::w32_regex_traits<char> > >(class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class boost::match_results<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class std::allocator<struct boost::sub_match<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > > > > > &,class boost::basic_regex<char,struct boost::regex_traits<char,class boost::w32_regex_traits<char> > > const &,enum boost::regex_constants::_match_flags)" (??$regex_match@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@V?$allocator@U?$sub_match@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@@boost@@@2@DU?$regex_traits@DV?$w32_regex_traits@D@boost@@@boost@@@boost@@YA_NV?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@0AEAV?$match_results@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@V?$allocator@U?$sub_match@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@@boost@@@2@@0@AEBV?$basic_regex@DU?$regex_traits@DV?$w32_regex_traits@D@boost@@@boost@@@0@W4_match_flags@regex_constants@0@@Z)
WorldState.cc.obj : error LNK2001: unresolved external symbol "private: void __cdecl boost::re_detail_107500::perl_matcher<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > >,class std::allocator<struct boost::sub_match<class std::_String_const_iterator<class std::_String_val<struct std::_Simple_types<char> > > > >,struct boost::regex_traits<char,class boost::w32_regex_traits<char> > >::construct_init(class boost::basic_regex<char,struct boost::regex_traits<char,class boost::w32_regex_traits<char> > > const &,enum boost::regex_constants::_match_flags)" (?construct_init@?$perl_matcher@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@V?$allocator@U?$sub_match@V?$_String_const_iterator@V?$_String_val@U?$_Simple_types@D@std@@@std@@@std@@@boost@@@2@U?$regex_traits@DV?$w32_regex_traits@D@boost@@@boost@@@re_detail_107500@boost@@AEAAXAEBV?$basic_regex@DU?$regex_traits@DV?$w32_regex_traits@D@boost@@@boost@@@3@W4_match_flags@regex_constants@3@@Z)
gazebo_physics.dll : fatal error LNK1120: 2 unresolved externals
~~~

Blast545, 2022-02-10:
:man_farmer: This error started appearing in the buildfarm install jobs: https://build.osrfoundation.org/job/gazebo-ci-gazebo11-windows7-amd64/122/console

j-rivero, 2022-02-10:
```
[  7%] Building CXX object gazebo/common/CMakeFiles/gazebo_common.dir/Assert.cc.obj
C:\vcpkg\installed\x64-windows\include\boost/detail/sp_typeinfo.hpp(23): note: This header is deprecated. Use <boost/core/typeinfo.hpp> instead.
```
There are vcpkg files being used in that build. This should not happen. I'm working on a fix for this which could be the root cause of the problem.

closed 2022-02-15 by scpeters