gazebosim/gazebo-classic #3173: Services calls [/shadow_caster_material_name] and  [/shadow_caster_render_back_faces] timed out

ljaniec, 2022-02-11:
Setup: Ubuntu 20.04, ROS 2 Galactic, Gazebo (11.10.1-1~focal)

With the update to the `gazebo11` (ver 11.10.1)  in the last ROS 2 Galactic sync my launch files and Gazebo overall stopped working correctly (services are timed out).

```
dpkg -l | grep gazebo
ii  gazebo11                                            11.10.1-1~focal                           amd64        Open Source Robotics Simulator
ii  gazebo11-common                                     11.10.1-1~focal                           all          Open Source Robotics Simulator - Shared files
ii  gazebo11-plugin-base                                11.10.1-1~focal                           amd64        Open Source Robotics Simulator - base plug-ins
ii  libgazebo11:amd64                                   11.10.1-1~focal                           amd64        Open Source Robotics Simulator - shared library
ii  libgazebo11-dev:amd64                               11.10.1-1~focal                           amd64        Open Source Robotics Simulator - Development Files
ii  ros-galactic-gazebo-dev                             3.5.2-5focal.20220124.195852              amd64        Provides a cmake config for the default version of Gazebo for the ROS distribution.
ii  ros-galactic-gazebo-msgs                            3.5.2-5focal.20220202.225637              amd64        Message and service data structures for interacting with Gazebo from ROS2.
ii  ros-galactic-gazebo-plugins                         3.5.2-5focal.20220203.025751              amd64        Robot-independent Gazebo plugins for sensors, motors and dynamic reconfigurable components.
ii  ros-galactic-gazebo-ros                             3.5.2-5focal.20220203.021749              amd64        Utilities to interface with Gazebo through ROS.
ii  ros-galactic-gazebo-ros-pkgs                        3.5.2-5focal.20220203.035839              amd64        Interface for using ROS with the Gazebo simulator.
ii  ros-galactic-nav2-gazebo-spawner                    1.0.8-1focal.20220203.014643              amd64        Package for spawning a robot model into Gazebo for Nav2
ii  ros-galactic-turtlebot3-gazebo                      2.2.4-2focal.20220203.040439              amd64        Gazebo simulation package for the TurtleBot3
ii  ros-galactic-turtlebot3-gazebo-dbgsym               2.2.4-2focal.20220203.040439              amd64        debug symbols for ros-galactic-turtlebot3-gazebo
```

After starting `gazebo --verbose` I got:

```
~$ gazebo --verbose
Gazebo multi-robot simulator, version 11.10.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.10.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.10.103
[Msg] Loading world file [/usr/share/gazebo-11/worlds/empty.world]
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.10.103
[Err] [Scene.cc:227] Service call[/shadow_caster_material_name] timed out
[Err] [Scene.cc:249] Service call[/shadow_caster_render_back_faces] timed out
[Wrn] [Scene.cc:462] Ignition transport [/scene_info] service call failed, falling back to gazebo transport [scene_info] request.
```
 
Part of the terminal logs after using my simulation launch file:

```
[nav2_gazebo_spawner-2] [INFO] [1644595446.406743712] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[nav2_gazebo_spawner-3] [INFO] [1644595446.462208129] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[nav2_gazebo_spawner-4] [INFO] [1644595446.469925262] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[gazebo-1] Gazebo multi-robot simulator, version 11.10.1
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[nav2_gazebo_spawner-2] [INFO] [1644595446.663426899] [entity_spawner]: Connecting to `/spawn_entity` service...
[gazebo-1] Gazebo multi-robot simulator, version 11.10.1
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[nav2_gazebo_spawner-3] [INFO] [1644595446.739736214] [entity_spawner]: Connecting to `/spawn_entity` service...
[nav2_gazebo_spawner-4] [INFO] [1644595446.744334475] [entity_spawner]: Connecting to `/spawn_entity` service...
[rviz2-16] [INFO] [1644595448.921861782] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-16] [INFO] [1644595448.922153597] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-16] [INFO] [1644595449.036399848] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1644595449.040512526] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1644595449.040642670] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-5] [INFO] [1644595449.157253467] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-27] [INFO] [1644595449.219489985] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-27] [INFO] [1644595449.219624549] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-27] [INFO] [1644595449.314909029] [rviz2]: Stereo is NOT SUPPORTED
[gazebo-1] [Msg] Waiting for master.
[gazebo-1] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gazebo-1] [Msg] Publicized address: 192.168.10.103
[gazebo-1] [Err] [Scene.cc:227] Service call[/shadow_caster_material_name] timed out
[gazebo-1] [Err] [Scene.cc:249] Service call[/shadow_caster_render_back_faces] timed out
[gazebo-1] [Wrn] [Scene.cc:462] Ignition transport [/scene_info] service call failed, falling back to gazebo transport [scene_info] request.
```
But I could put the robot manually in the world, with this output:

```
[gazebo-1] [INFO] [1644595511.485998038] [turtlebot3_imu]: <initial_orientation_as_reference> is unset, using default value of false to comply with REP 145 (world as orientation reference)
[gazebo-1] [INFO] [1644595511.786551485] [turtlebot3_diff_drive]: Wheel pair 1 separation set to [0.160000m]
[gazebo-1] [INFO] [1644595511.786631388] [turtlebot3_diff_drive]: Wheel pair 1 diameter set to [0.066000m]
[gazebo-1] [INFO] [1644595511.788092149] [turtlebot3_diff_drive]: Subscribed to [/cmd_vel]
[gazebo-1] [INFO] [1644595511.790290199] [turtlebot3_diff_drive]: Advertise odometry on [/odom]
[gazebo-1] [INFO] [1644595511.792370222] [turtlebot3_diff_drive]: Publishing odom transforms between [odom] and [base_footprint]
[gazebo-1] [INFO] [1644595511.810008709] [turtlebot3_joint_state]: Going to publish joint [wheel_left_joint]
[gazebo-1] [INFO] [1644595511.810074926] [turtlebot3_joint_state]: Going to publish joint [wheel_right_joint]
```

Reverting to the old version of Gazebo packages didn't help, but I am not sure if I did it correctly too (maybe I have missed some dependent ROS packages). I have used:
```
sudo apt-get install gazebo11=11.9.0-1~focal gazebo11-common=11.9.0-1~focal gazebo11-plugin-base=11.9.0-1~focal libgazebo11=11.9.0-1~focal libgazebo11-dev=11.9.0-1~focal 
```

Launcher

scpeters, 2022-02-14:
it sounds like ignition transport services are not working properly, and this causes gazebo to fail to load? Do you see anything in `gzclient`? It should still attempt to load the scene using gazebo-transport

ljaniec, 2022-02-14:
[image]

Gazebo starts with `empty.world` scene, but any launcher (mines or from Nav2 repository and `nav2_bringup` package) cannot call any services (in particular `spawn_entity`).
Example:
`ljaniec@ljaniec-PC:~/ros2_ws$ ros2 launch nav2_bringup multi_tb3_simulation_launch.py` gives:
[image]

relffok, 2022-02-15:
> it sounds like ignition transport services are not working properly, and this causes gazebo to fail to load? Do you see anything in gzclient? It should still attempt to load the scene using gazebo-transport

Gazebo is loading up and you can see the scene and even use it, though the services are not available. You can also reproduce this by just running the empty world using `gazebo --verbose`.

ljaniec, 2022-03-04:
I have found something unusual and weird, which maybe could help a bit there - I can use VirtualBox with Ubuntu 20.04 and ROS2 Galactic (binaries from Debian packages) with Gazebo without these errors of timed out services... 

What is the source of this really weird situation? What do I have to change in my setup for everything to work? 

Will building Gazebo from source help? It's last thing I haven't tried yet...

ljaniec, 2022-03-15:
@relffok @scpeters 
There are some new details found by @beezow:

https://githubhot.com/repo/lopsided98/nix-ros-overlay/issues/161
> 
> So while gazebo was working, when trying to use gazebo_ros there were still some problems.
> 
> ➤ rosrun gazebo_ros gzserver -e ode worlds/emtpy.world --verbose $final
> Gazebo multi-robot simulator, version 11.9.0
> Copyright (C) 2012 Open Source Robotics Foundation.
> Released under the Apache 2 License.
> http://gazebosim.org
> 
> munmap_chunk(): invalid pointer
> /nix/store/8fj7m50ag8si1gzgdya7j2v2b9r5z1dw-ros-noetic-gazebo-ros-2.9.2-r1/lib/gazebo_ros/gzserver: line 41: 179679 Aborted                 (core dumped) GAZEBO_MASTER_URI="$desired_master_uri" GAZEBO_MODEL_DATABASE_URI="$desired_model_database_uri" gzserver $final
> 
> I was able to track this down the libgazebo_ros_paths_plugin.so as triggering the fault. When removed from the shared library list for gzserver, the error goes away. I tested this by directly calling gzserver with the args from gazebo_ros. This can be found in /nix/store/8fj7m50ag8si1gzgdya7j2v2b9r5z1dw-ros-noetic-gazebo-ros-2.9.2-r1/lib/gazebo_ros.
> 
> gzserver -e ode worlds/empty.world --verbose -s /nix/store/8fj7m50ag8si1gzgdya7j2v2b9r5z1dw-ros-noetic-gazebo-ros-2.9.2-r1/lib/libgazebo_ros_paths_plugin.so -s /nix/store/8fj7m50ag8si1gzgdya7j2v2b9r5z1dw-ros-noetic-gazebo-ros-2.9.2-r1/lib/libgazebo_ros_api_plugin.so
> 
> I dont know if this is related or not but wanted to document my efforts to get gazebo working again
> 

and

> After doing some debugging, `ros::package::getPlugins("gazebo_ros","gazebo_media_path",gazebo_media_paths);` in libgazebo_ros_paths_plugin.cpp is causing problems. In fact, every call to `ros::package::getPlugins` will trigger the fault. 
> (...)
> If anyone has any insights on what may be causing the issue that would be great. I notice the call stack goes through boost so I wonder if somehow multiple versions are being pulled in? Or maybe something with the rospack dependencies are mucked up?

and

> It seems gazebo was pulling in boost 173, while gazebo_ros was pulling in boost 177. By upgrading gazebo to boost177, the segfaults went away. Can finally run gazebo_ros again! You should be able to use the staging branch as it incorporates all the updates needed.
> 
> Can someone else can confirm the staging branch works for gazebo_ros?
> 
> The only remaining problem is with the shadows. There are some errors related to the shadow caster. The shadows from light sources do not work, with the light shining right through any obstacles.
>

ljaniec, 2022-03-17:
I got some more warnings, maybe they will help:
```
[Wrn] [Scene.cc:462] Ignition transport [/scene_info] service call failed, falling back to gazebo transport [scene_info] request.
```
After 1 TurtleBot3 Waffle is added to the empty world and clicking Window  > Plot, I can generate this errors:
```
[Err] [Palette.cc:783] No introspection managers detected.
[Err] [Palette.cc:784] Is a gzserver running?
[Err] [IntrospectionCurveHandler.cc:224] No introspection managers detected.
```
There are also no ROS2 topics coming from the robot, only:
```
ljaniec@ljaniec-PC:~$ ros2 topic list
/parameter_events
/rosout
```

ljaniec, 2022-03-21:
There was an update to the Gazebo 11 and by struck of luck I have found out what was maybe not the problem, but at least a solution - I had to reset my Firewall settings in Ubuntu 20.04 because even `ros2 topic list` wasn't working correctly. Then I tried to launch `gazebo --verbose` and it didn't hanged, I could add TB3 to the empty world and see it's topics etc.

If this is acceptable solution to this problem, I think this issue could be closed.

mengchaoheng, 2022-04-28:
@ljaniec Disconnect from the internet and no more errors.

scpeters, 2022-04-28:
can you try setting `IGN_IP=127.0.0.1`?

ljaniec, 2022-05-10:
I cannot reproduce this error in a meaningful way right now to check it, sorry.

I will close this issue with these steps written down for future users

closed 2022-05-10 by ljaniec

mengchaoheng, 2022-05-10:
> I cannot reproduce this error in a meaningful way right now to check it, sorry.
> 
> 
> 
> I will close this issue with these steps written down for future users

F*** you

cambel, 2025-01-31:
I encountered this issue today on my docker environment with ROS1 Noetic and Gazebo 11. 

> can you try setting IGN_IP=127.0.0.1 as an environment variable?

Setting this as env variable worked and the issue is resolved for me.