gazebosim/gazebo-classic #3176: Linker errors using GAZEBO_LIBRARIES

relffok, 2022-02-15:
I have a clean Ubuntu 20.04 setup with a fresh ros galactic and gazebo(debian 11.10.1)  installation.
When I try to build my ros2 workspace (for example `turtlebot_simulations` package) and link against `$GAZEBO_LIBRARIES`, I get  the error concerning dart include path. 
```
CMake Error in CMakeLists.txt:
  Imported target "dart" includes non-existent path
    "/include"
  in its INTERFACE_INCLUDE_DIRECTORIES. 
```

I wanted to check the libraries, and printed the output of the variable `$GAZEBO_LIBRARIES`, which was: 

`BulletSoftBodyBulletDynamicsBulletCollisionLinearMathSimTKcommonSimTKmathSimTKsimbodydartdart/usr/lib/x86_64-linux-gnu/libgazebo.so/usr/lib/x86_64-linux-gnu/libgazebo_client.so/usr/lib/x86_64-linux-gnu/libgazebo_gui.so/usr/lib/x86_64-linux-gnu/libgazebo_sensors.so/usr/lib/x86_64-linux-gnu/libgazebo_rendering.so/usr/lib/x86_64-linux-gnu/libgazebo_physics.so/usr/lib/x86_64-linux-gnu/libgazebo_ode.so/usr/lib/x86_64-linux-gnu/libgazebo_transport.so/usr/lib/x86_64-linux-gnu/libgazebo_msgs.so/usr/lib/x86_64-linux-gnu/libgazebo_util.so/usr/lib/x86_64-linux-gnu/libgazebo_common.so/usr/lib/x86_64-linux-gnu/libgazebo_gimpact.so/usr/lib/x86_64-linux-gnu/libgazebo_opcode.so/usr/lib/x86_64-linux-gnu/libgazebo_opende_ou.soBoost::threadBoost::systemBoost::filesystemBoost::program_optionsBoost::regexBoost::iostreamsBoost::date_time/usr/lib/x86_64-linux-gnu/libprotobuf.so-lpthreadsdformat9::sdformat9optimized/usr/lib/x86_64-linux-gnu/libOgreMain.sodebug/usr/lib/x86_64-linux-gnu/libOgreMain.soBoost::threadBoost::date_timeoptimized/usr/lib/x86_64-linux-gnu/libOgreTerrain.sodebug/usr/lib/x86_64-linux-gnu/libOgreTerrain.sooptimized/usr/lib/x86_64-linux-gnu/libOgrePaging.sodebug/usr/lib/x86_64-linux-gnu/libOgrePaging.soignition-math6::requestedignition-transport8::requestedignition-msgs5::requestedignition-common3::ignition-common3-graphicsignition-fuel_tools4::requested`

I could not find any issue related to this on dartsim. Am I missing something here or is this related to the latest update?

shrinivas96, 2023-03-02:
What helped you solve your issue? 

I have a similar error while building `turtlebot3_gazebo` on ROS noetic. Everything was working fine, and then I had to clean my workspace and build again, when I started getting the above error. To be more specific, the build fails and gives this error message: 

```
CMake Error in CMakeLists.txt:
  Imported target "dart" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

Any help is appreciated.

traversaro, 2023-03-03:
I would suggest to post on Gazebo Answers, it is difficult to track questions if they are asked on closed GitHub issues.