gazebosim/gazebo-classic #3177: Compiling on `g++-11` with `-std=c++20`

Ondraceq, 2022-02-15:
The gazebo header files won't compile with `g++-11` with the `-std=c++20` flag.

It seems like gcc decided to change something in a weird way (it works with g++-10 and std20 - or - with g++-11 and std17), but when I looked at the errors, it shows:
```
/usr/include/gazebo-9/gazebo/common/MovingWindowFilter.hh:48:44: error: expected unqualified-id before ‘)’ token
   48 |       public: MovingWindowFilterPrivate<T>();
      |                                            ^
/usr/include/gazebo-9/gazebo/common/MovingWindowFilter.hh:71:5: error: no declaration matches ‘gazebo::common::MovingWindowFilterPrivate<T>::MovingWindowFilterPrivate()’
   71 |     MovingWindowFilterPrivate<T>::MovingWindowFilterPrivate() :
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-9/gazebo/common/MovingWindowFilter.hh:71:5: note: no functions named ‘gazebo::common::MovingWindowFilterPrivate<T>::MovingWindowFilterPrivate()’
/usr/include/gazebo-9/gazebo/common/MovingWindowFilter.hh:45:11: note: ‘class gazebo::common::MovingWindowFilterPrivate<T>’ defined here
   45 |     class MovingWindowFilterPrivate
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-9/gazebo/common/MovingWindowFilter.hh:86:37: error: expected unqualified-id before ‘)’ token
   86 |       public: MovingWindowFilter<T>();
      |                                     ^
/usr/include/gazebo-9/gazebo/common/MovingWindowFilter.hh:124:5: error: no declaration matches ‘gazebo::common::MovingWindowFilter<T>::MovingWindowFilter()’
  124 |     MovingWindowFilter<T>::MovingWindowFilter()
      |     ^~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-9/gazebo/common/MovingWindowFilter.hh:113:27: note: candidate is: ‘gazebo::common::MovingWindowFilter<T>::MovingWindowFilter(gazebo::common::MovingWindowFilterPrivate<T>&)’
  113 |       protected: explicit MovingWindowFilter<T>(
      |                           ^~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-9/gazebo/common/MovingWindowFilter.hh:83:11: note: ‘class gazebo::common::MovingWindowFilter<T>’ defined here
   83 |     class MovingWindowFilter
      |           ^~~~~~~~~~~~~~~~~~
```
I know that it shows `gazebo-9`, but I checked as well that the same problem is with the current version in `gazebo11` (at d4dbf98).

The problem as I understand it is that a templated class cannot have the template parameter repeated in the constructor - this was never allowed in c++.

### Suggested fix
Remove the `<T>` from the constructors when the constructors themselves are not templated.

WBurzynski, 2022-07-13:
i run into the same issue. Were there any updates?

WBurzynski, 2022-07-13:
I tried to implement fix myself, but i seems there is way more work to do than just removing `<T>` eg. lot of stuff was removed in `std::allocator`.

Ondraceq, 2023-02-03:
To be clear, the issue is addressing compilation of other projects that include the gazebo headers. For this, the removal of `<T>` is sufficient (right now we have to hack it by adding the same file into include path before gazebo include paths).

traversaro, 2023-02-03:
Some of the `<T>` were added in https://github.com/gazebosim/gazebo-classic/commit/21d233ff39d2c9e91fc7509620a06a5969695f73#diff-08e12ed58d000f7255113fa121fabb2db3b33836f4a001228300bef26420f1ddR112, so I guess they were related to some cppcheck check. Anyhow @Ondraceq if you like you can try to submit a PR to fix the problem, and then we can see if the CI jobs are passing or not.

PR #3295 (2023-02-05): Fix template specialization in constructors
# 🦟 Bug fix

Fixes #3177

## Summary
Just removing `<T>` where there is no `template` before the function. (Somehow gcc didn't mind in version 10, but now correctly minds in version 11.
commit 08f01aa8 (2023-02-05): Fix template specialization in constructors (#3295)

Fixes #3177.
closed 2023-02-05 by scpeters