gazebosim/gazebo-classic #3180: Fails to build with ffmpeg 5.0

sebastinas, 2022-02-17:
gazebo is not compatible with ffmpeg 5.0:
```
/<<PKGBUILDDIR>>/gazebo/common/AudioDecoder.cc: In member function ‘bool gazebo::common::AudioDecoder::Decode(uint8_t**, unsigned int*)’:                      
/<<PKGBUILDDIR>>/gazebo/common/AudioDecoder.cc:100:17: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]                     
  100 |   av_init_packet(&packet);                                                                                                                             
      |   ~~~~~~~~~~~~~~^~~~~~~~~                                                                                                                              
In file included from /usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:45,                                                                                   
                 from /<<PKGBUILDDIR>>/gazebo/common/ffmpeg_inc.h:32,                                                                                          
                 from /<<PKGBUILDDIR>>/gazebo/common/AudioDecoder.cc:21:                                                                                       
/usr/include/x86_64-linux-gnu/libavcodec/packet.h:506:6: note: declared here                                                                                   
  506 | void av_init_packet(AVPacket *pkt);                                                                                                                    
      |      ^~~~~~~~~~~~~~                                                                                                                                    
/<<PKGBUILDDIR>>/gazebo/common/AudioDecoder.cc:116:24: error: ‘avcodec_decode_audio4’ was not declared in this scope; did you mean ‘avcodec_decode_subtitle2’? 
  116 |         bytesDecoded = avcodec_decode_audio4(this->codecCtx, decodedFrame,                                                                             
      |                        ^~~~~~~~~~~~~~~~~~~~~                                                                                                           
      |                        avcodec_decode_subtitle2                                                                                                        
/<<PKGBUILDDIR>>/gazebo/common/AudioDecoder.cc: In member function ‘bool gazebo::common::AudioDecoder::SetFile(const string&)’:                                
/<<PKGBUILDDIR>>/gazebo/common/AudioDecoder.cc:217:38: error: ‘AVStream’ {aka ‘struct AVStream’} has no member named ‘codec’                                   
  217 |     if (this->formatCtx->streams[i]->codec->codec_type == AVMEDIA_TYPE_AUDIO)                                                                          
      |                                      ^~~~~                                                                                                             
/<<PKGBUILDDIR>>/gazebo/common/AudioDecoder.cc:241:59: error: ‘AVStream’ {aka ‘struct AVStream’} has no member named ‘codec’                                   
  241 |   this->codecCtx = this->formatCtx->streams[audioStream]->codec;                                                                                       
      |                                                           ^~~~~                                                                                        
/<<PKGBUILDDIR>>/gazebo/common/AudioDecoder.cc:247:37: error: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]                            
  247 |   this->codec = avcodec_find_decoder(codecCtx->codec_id);                                                                                              
      |                 ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~                                                                                               
      |                                     |                                                                                                                  
      |                                     const AVCodec*      
```

traversaro, 2022-02-27:
Related issue (it was on the deprecated function that eventually were removed): https://github.com/osrf/gazebo/issues/2002

traversaro, 2022-02-27:
A bit of info on replacements extracted from ffmpeg: 
* `avcodec_decode_audio4` needs to be substitued with Use [avcodec_send_packet()](http://ffmpeg.org/doxygen/4.1/group__lavc__decoding.html#ga58bc4bf1e0ac59e27362597e467efff3) and [avcodec_receive_frame()](http://ffmpeg.org/doxygen/4.1/group__lavc__decoding.html#ga11e6542c4e66d3028668788a1a74217c). (http://ffmpeg.org/doxygen/4.1/group__lavc__decoding.html#gaaa1fbe477c04455cdc7a994090100db4)
* The `codec` member of `AVStream` needs to be substituted with "the codecpar struct instead" (https://github.com/FFmpeg/FFmpeg/blob/n4.4.1/libavformat/avformat.h#L885)

traversaro, 2022-02-27:
For the usual `#if` preprocessor magic, in `libavformat/version.h` the macro `LIBAVFORMAT_VERSION_MAJOR` is defined, that is:
* 58 for ffmpeg v4.4.1 : https://github.com/FFmpeg/FFmpeg/blob/n4.4.1/libavformat/version.h#L34
* 59 for ffmpeg v5.0 : https://github.com/FFmpeg/FFmpeg/blob/n4.4.1/libavformat/version.h#L34

traversaro, 2022-02-27:
Strictly related issue: https://github.com/ignitionrobotics/ign-common/issues/306 (as the affected classes are the same that as some point were copied there).

KenwoodFox, 2022-03-08:
I also have this issue on manjaro linux

vebjornjr, 2022-03-10:
@KenwoodFox, did you manage to fix it? I experience the same on Manjaro when trying to build ros-noetic-desktop-full. I don't understand how to fix this myself, @traversaro.

traversaro, 2022-03-10:
Basically we need to substitute the function that were removed from ffmpeg API with the one that are still existing, see https://github.com/osrf/gazebo/issues/3180#issuecomment-1053641133 . If you want to try to fix it, perhaps it is a good idea to first get a basic idea of how the ffmpeg API works, for example reading the examples in https://github.com/FFmpeg/FFmpeg/tree/870bfe16a12bf09dca3a4ae27ef6f81a2de80c40/doc/examples . Once you got the general idea, I guess it may be easier to fix the code in Gazebo. Furthermore, a possible easier stepping stone may be fixing ffmpeg support in https://github.com/ignitionrobotics/ign-common/issues/306 .

KenwoodFox, 2022-03-11:
> @KenwoodFox, did you manage to fix it? I experience the same on Manjaro when trying to build ros-noetic-desktop-full. I don't understand how to fix this myself, @traversaro.

I have not yet, I fail building ignition due to ffmpeg deps, this may take a PR to solve I'll look into learning the API

JorgeEdGG, 2022-03-11:
@vebjornjr @traversaro @KenwoodFox 
Hi guys, the workaround that work for me was first unistall ffmpeg then install gazebo using ffmpeg4.4, finaly reinstall ffmpeg, i hope it works for you.

sudo pacman -Rd --nodeps ffmpeg (and any other ffmpeg you have installed, like the git version)
sudo pacman -S ffmpeg4.4
yay -S gazebo
sudo pacman -S ffmpeg

KenwoodFox, 2022-03-11:
> @vebjornjr @traversaro @KenwoodFox Hi guys, the workaround that work for me was first unistall ffmpeg then install gazebo using ffmpeg4.4, finaly reinstall ffmpeg, i hope it works for you.
>
> sudo [...]

This worked!

vebjornjr, 2022-03-12:
> @vebjornjr @traversaro @KenwoodFox Hi guys, the workaround that work for me was first unistall ffmpeg then install gazebo using ffmpeg4.4, finaly reinstall ffmpeg, i hope it works for you.
>
> sudo [...]

Thank you, it worked for me too.

traversaro, 2022-03-19:
> * `avcodec_decode_audio4` and `avcodec_decode_video2` need to be substitued with Use [avcodec_send_packet()](http://ffmpeg.org/doxygen/4.1/group__lavc__decoding.html#ga58bc4bf1e0ac59e27362597e467efff3) and [avcodec_receive_frame()](http://ffmpeg.org/doxygen/4.1/group__lavc__decoding.html#ga11e6542c4e66d3028668788a1a74217c). (http://ffmpeg.org/doxygen/4.1/group__lavc__decoding.html#gaaa1fbe477c04455cdc7a994090100db4 and http://ffmpeg.org/doxygen/4.1/group__lavc__decoding.html#ga3ac51525b7ad8bca4ced9f3446e96532)

"Need to be substituted" is a bit generic. I found the relevant commit that change the ffmpeg examples to understand how they can be substituted:
* https://github.com/FFmpeg/FFmpeg/commit/3d66717f7cb5555257244be8f5bce172ed3af7ac

traversaro, 2022-03-19:
For `avcodec_decode_video2` this ign-common commit is strictly related: https://github.com/ignitionrobotics/ign-common/commit/e2f6a84100f39fe29d5ffc49447e12c1f6c4196d .

traversaro, 2022-03-19:
I guess that backporting:
* https://github.com/ignitionrobotics/ign-common/commit/e2f6a84100f39fe29d5ffc49447e12c1f6c4196d
* https://github.com/ignitionrobotics/ign-common/pull/325

Should fix the problem also for Classic Gazebo.

traversaro, 2022-04-19:
A fix for this issue is provided in https://github.com/osrf/gazebo/pull/3195 .

scpeters, 2022-06-09:
Fixed by https://github.com/osrf/gazebo/pull/3195.

closed 2022-06-09 by scpeters