gazebosim/gazebo-classic #3189: SetAxis function replacement for gazebo11

elle-miller, 2022-03-15:
Hi,

When simulating a mobile robot in Gazebo 11 using controller code written for Gazebo 9, I am noticing funny motion behaviour e.g. velocity controller works perfectly in x direction but not for y or z.

I use the function 'SetAxis' from /gazebo/physics in my code, which I noticed was deprecated by @chapulina in commit `5a0c754589e406d0b0104fd12edaa1430c72d1ab`.

If my previous code was:

passive_joint_->SetAxis(0, axis_);

What should I now include/replace it with?

Thank you!
Elle

j-rivero, 2022-03-15:
Hello Elle. We tried to document all the required changes for deprecated methods in https://github.com/osrf/gazebo/blob/gazebo11/Migration.md. Looking at the commit you mention seems like the deprecation cycle started in gazebo8. Anyway, You can check the API changed documented if you search the Changelog document by `SetAxis`. Hope it helps.

elle-miller, 2022-05-02:
@j-rivero

Hi Jose, thanks for the help.

Unfortunately I have been unable to track down the replacement function for **SetAxis** in the Changelog. The only mention of the function is for Gazebo 7->8 when `math` was updated to `ignition::math`.

I see that the function ```ResolveAxisXyz``` was added in last commit f946cb3edbft (2020) "add helper function for resolving joint axis, pass resolved frame to Joint::SetAxis to simplify engine-specific code", but this doesn't make sense to me as I thought SetAxis was deprecated in @chapulina 's commit 5a0c75458 (2017) as mentioned in initial post.

`this->SetAxis(0, this->ResolveAxisXyz(0));` (physics/Joint.cc line 380)

I have searched through the commits around the deprecation but it remains unclear to me what was replaced (and what I should use now). If you have any further advice or help to offer that would be greatly appreciated - cheers! :)

j-rivero, 2022-05-02:
> Hi Jose, thanks for the help.

You are welcome.

> Unfortunately I have been unable to track down the replacement function for **SetAxis** in the Changelog. The only mention of the function is for Gazebo 7->8 when `math` was updated to `ignition::math`.

I think that is exactly was is being done in https://github.com/osrf/gazebo/commit/5a0c754589e406d0b0104fd12edaa1430c72d1ab

> I have searched through the commits around the deprecation but it remains unclear to me what was replaced (and what I should use now). If you have any further advice or help to offer that would be greatly appreciated

From the Changelog:
```
Deprecation: void SetAxis(unsigned int _index, const math::Vector3 &_axis) = 0
Replacement: void SetAxis(const unsigned int _index, const ignition::math::Vector3d &_axis) = 0
```
The function is now using ignition::math input types instead of native Gazebo code, that should be the change you need to make.