gazebosim/gazebo-classic #3198: Gazebo on Windows: conda-forge installation

cogneos, 2022-03-23:
After a seemingly successful installation, gazebo fails to launch w/ the following error message
[image]

traversaro, 2022-03-24:
Can you report the output of conda list command?

cogneos, 2022-03-29:
> Can you report the output of conda list command?

Here is goes:

# packages in environment at C:\Users\gorokhov\Programs\MiniConda:
#
# Name                    Version                   Build  Channel
blosc                     1.21.0               h0e60522_0    conda-forge
boost-cpp                 1.74.0               h9f4b32c_7    conda-forge
brotlipy                  0.7.0           py39h2bbff1b_1003  
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cairo                     1.16.0            h15b3021_1010    conda-forge
certifi                   2021.10.8        py39hcbf5309_1    conda-forge
cffi                      1.15.0           py39h2bbff1b_0  
cfitsio                   4.0.0                hd67004f_0    conda-forge
charset-normalizer        2.0.4              pyhd3eb1b0_0  
conda                     4.12.0           py39hcbf5309_0    conda-forge
conda-content-trust       0.1.1              pyhd3eb1b0_0  
conda-package-handling    1.7.3            py39h8cc25b3_1  
console_bridge            1.0.2                h5362a0b_0    conda-forge
console_shortcut          0.1.1                         4  
cppzmq                    4.7.1                h50e1b43_1    conda-forge
cryptography              36.0.0           py39h21b164f_0  
curl                      7.82.0               h789b8ee_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.4.7                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              hce3cb01_2    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
gazebo                    11.10.2              h6d7d6d0_0    conda-forge
geos                      3.10.2               h39d44d4_0    conda-forge
geotiff                   1.7.0                h38b14a8_7    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.12.1          nompi_h2a0e4a3_104    conda-forge
icu                       69.1                 h0e60522_0    conda-forge
idna                      3.3                pyhd3eb1b0_0  
ilmbase                   2.5.5                h12d4b20_0    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jpeg                      9e                   h8ffe710_0    conda-forge
jsoncpp                   1.9.5                h2d74725_1    conda-forge
jxrlib                    1.1                  h8ffe710_2    conda-forge
kealib                    1.4.14               h8995ca9_3    conda-forge
krb5                      1.19.3               h1176d77_0    conda-forge
lcms2                     2.12                 h2a16943_0    conda-forge
lerc                      3.0                  h0e60522_0    conda-forge
libclang                  13.0.1          default_h81446c8_0    conda-forge
libcurl                   7.82.0               h789b8ee_0    conda-forge
libdeflate                1.10                 h8ffe710_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libgdal                   3.4.2                h7f66a12_0    conda-forge
libglib                   2.70.2               h3be07f2_4    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
libignition-cmake2        2.11.0               h0e60522_0    conda-forge
libignition-common3       3.13.2               h9bd54ec_1    conda-forge
libignition-fuel-tools4   4.4.0                h98d6f7e_5    conda-forge
libignition-math6         6.10.0               h0e60522_0    conda-forge
[... 55 lines cut ...]
six                       1.16.0             pyhd3eb1b0_0  
sqlite                    3.37.0               h2bbff1b_0  
tbb                       2021.5.0             h2d74725_0    conda-forge
tbb-devel                 2021.5.0             h2d74725_0    conda-forge
tiledb                    2.7.1                h95dad36_0    conda-forge
tiny-process-library      2.0.4                h0e60522_0    conda-forge
tinyxml                   2.6.2                h2d74725_2    conda-forge
tinyxml2                  9.0.0                h0e60522_2    conda-forge
tqdm                      4.62.3             pyhd3eb1b0_1  
tzdata                    2021e                hda174b7_0  
urdfdom                   2.3.5                h2d74725_0    conda-forge
[... cut ...]
wincertstore              0.2              py39haa95532_2  
xerces-c                  3.2.3                h0e60522_4    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                he774522_0  
zeromq                    4.3.3                h0e60522_3    conda-forge
zlib                      1.2.11            h8ffe710_1014    conda-forge
zstd                      1.5.2                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge

traversaro, 2022-03-29:
Related reply: https://github.com/conda-forge/gazebo-feedstock/issues/125#issuecomment-1081560181 . 

TL;DR: Have you tried to install gazebo in a new clean environment (i.e. `mamba create -n gazebo -c conda-forge gazebo`) instead of the `base` environment and check if the problem is still there? Thanks!

cogneos, 2022-03-29:
It looks like the installation succeded now. Thank you!

traversaro, 2022-03-29:
Great!

closed 2022-04-03 by cogneos