gazebosim/gazebo-classic #3202: Gazebo on Windows: vcpkg installation

j-rivero, 2022-03-24:
The [vcpkg](https://github.com/microsoft/vcpkg) project has available a port for [Gazebo11](https://github.com/microsoft/vcpkg/tree/master/ports/gazebo). The Goal of the ticket is to evaluate if this installation is stable enough to go into our documentation.

**Help wanted:**
We are interested in knowing if our users on Windows can install and use it without problems. If you want to help, please follow vcpkg instructions to bootstrap vcpkg, try to compile Gazebo and try to run a basic simulation on it. Please comment in the issue with details about Windows version, problems/not-problems found, graphic card drivers installed and other information you find useful. You are also welcome to comment and help with other people's problem.

**Windows specific:** check https://github.com/osrf/gazebo/issues/2901 for the list of configurations, problems or issues on Windows.

j-rivero, 2022-03-24:
First of all, thanks so much @Ace314159 for the work done to bring Gazebo to vcpkg. Really cool work.

I was able to use my Windows10 to install the Gazebo port without problems in the compilation of all the dependencies and gazebo itself (well, limiting the compilation threads for Gazebo using `VCPKG_MAX_CONCURRENCY` variable).

Playing with vcpkg directories (`buildtress`, `installed` and `packages`) I was wondering, what is the way of invoking Gazebo? Are there some instructions about how to run it?

Thanks!

Ace314159, 2022-03-25:
@j-rivero I'm glad you were able to build successfully with that minor tweak!

The `buildtrees` folder has intermediates used while building and can be [deleted](https://vcpkg.readthedocs.io/en/latest/about/faq/#how-can-i-remove-temporary-files). Both the `packages` and `installed` folder contain the same files, just organized differently. Due to how vcpkg builds are done, by default gazebo loads the built-in models from the `packages` folder, so both the `installed` and `packages` folders are recommended to keep.

In order to run gazebo, you need to set the `HOME` environment variable to your users directory (i.e. `C:\Users\{name}`). Then, you can just double click the `gazebo` executable in the `installed/x64-windows/tools/gazebo` folder.

You can also run the the executable through cmd or Powershell (with the HOME environment variable set), but I have noticed that using ctrl+c to kill it sometimes doesn't kill the `gzclient` and `gzserver` child processes. I have made https://github.com/osrf/gazebo/pull/3167 to attempt to fix this, and it seems to work for me.

traversaro, 2023-01-27:
I am not sure, but I think that https://github.com/microsoft/vcpkg/pull/27107 broke gazebo on vcpkg, at least when using the default version of the libraries at it bumped ogre from 1.12.9 to 13.4.4 . 

Indeed, I was quite curious about why the vcpkg CI did not catched the gazebo failure, and it seems that gazebo was skipped in CI as apparently VS 2022 Update 3 broke its compilation (see https://github.com/microsoft/vcpkg/pull/26283#issuecomment-1218636710) and so it was removed from the CI.

talregev, 2023-03-22:
I already did a PR that try to solve the issue there.
Also I revive it  recently. I can make a PR again, and help will apricate. 
Now day vcpkg support ogre 13.6.2 that was break the api of the previous one. 
Can you help me patch gazebo that it compile with the new ogre?

traversaro, 2023-03-22:
>  Can you help me patch gazebo that it will compile with the new ogre?

There is a relaed issue in https://github.com/gazebosim/gazebo-classic/issues/2700, unfortunatly I am afraid it will not be trivial.

talregev, 2023-03-23:
@traversaro I try to compile gazebo with the new ogre lib. But gazebo using private headers that vcpkg don't export.
I discuss it here
https://github.com/OGRECave/ogre/issues/2813
https://github.com/microsoft/vcpkg/issues/30376

Anybody can help me?

talregev, 2023-03-25:
I made a PR of gazebo in here. It still not fully compile, but It start to build. It need patch gazebo to fit the new ogre that vcpkg is using. 
https://github.com/microsoft/vcpkg/pull/30424

If you can help, please common on the vcpkg PR. Thank you.

talregev, 2023-04-05:
I made a PR that compile gazebo in windows.
https://github.com/gazebosim/gazebo-classic/pull/3312