gazebosim/gazebo-classic #3206: Gazebo in Windows 10: [Msg] Waiting for master.

cogneos, 2022-04-04:
Installed gazebo under Windows 10 last week and it worked for a few days. Starting yesterday, it doesn't start, snipped below:

(base) C:\Users\gorokhov>activate gazebo

(gazebo) C:\Users\gorokhov>gazebo --verbose
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Err] [D:\bld\gazebo_1647700150418\work\gazebo\transport\ConnectionManager.cc:121] Failed to connect to master in 30 seconds.
[Err] [D:\bld\gazebo_1647700150418\work\gazebo\gazebo_shared.cc:78] Unable to initialize transport.
[Err] [D:\bld\gazebo_1647700150418\work\gazebo\gazebo_client.cc:56] Unable to setup Gazebo

(gazebo) C:\Users\gorokhov>

kielerrr, 2022-04-19:
Try running `roscore`

closed 2022-07-03 by cogneos

mechyahya, 2024-06-28:
Did you solve this problem?

I got a same problem in ubuntu 20.04 and gazebo 11.14