gazebosim/gazebo-classic #3208: At RPi 4B  build gazebo-gazebo9_9.19.0 error: multiple definition of `typeinfo name for SingletonT<gazebo::common::SystemPaths>'

dongpan90, 2022-04-09:
**Environment：**
Linux pi-desktop 5.4.0-1058-raspi #65-Ubuntu SMP PREEMPT Fri Mar 25 12:34:25 UTC 2022 armv7l armv7l armv7l GNU/Linux

**Build package**: gazebo-gazebo9_9.19.0

**Build Error**:
`[ 18%] Linking CXX shared library libgazebo_common.so
/usr/bin/ld: CMakeFiles/gazebo_common.dir/ColladaLoader.cc.o:(.rodata+0x4): multiple definition of `typeinfo name for SingletonT<gazebo::common::SystemPaths>'; CMakeFiles/gazebo_common.dir/BVHLoader.cc.o:(.rodata+0x0): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/ColladaLoader.cc.o:(.data.rel.ro+0x0): multiple definition of `typeinfo for SingletonT<gazebo::common::SystemPaths>'; CMakeFiles/gazebo_common.dir/BVHLoader.cc.o:(.data.rel.ro+0x0): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/CommonIface.cc.o:(.rodata+0x0): multiple definition of `typeinfo name for SingletonT<gazebo::common::SystemPaths>'; CMakeFiles/gazebo_common.dir/BVHLoader.cc.o:(.rodata+0x0): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/CommonIface.cc.o:(.data.rel.ro+0x0): multiple definition of `typeinfo for SingletonT<gazebo::common::SystemPaths>'; CMakeFiles/gazebo_common.dir/BVHLoader.cc.o:(.data.rel.ro+0x0): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/Material.cc.o:(.rodata+0x0): multiple definition of `typeinfo name for SingletonT<gazebo::common::SystemPaths>'; CMakeFiles/gazebo_common.dir/BVHLoader.cc.o:(.rodata+0x0): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/Material.cc.o:(.data.rel.ro+0x0): multiple definition of `typeinfo for SingletonT<gazebo::common::SystemPaths>'; CMakeFiles/gazebo_common.dir/BVHLoader.cc.o:(.data.rel.ro+0x0): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/ModelDatabase.cc.o:(.rodata+0x4): multiple definition of `typeinfo name for SingletonT<gazebo::common::SystemPaths>'; CMakeFiles/gazebo_common.dir/BVHLoader.cc.o:(.rodata+0x0): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/ModelDatabase.cc.o:(.data.rel.ro+0x0): multiple definition of `typeinfo for SingletonT<gazebo::common::SystemPaths>'; CMakeFiles/gazebo_common.dir/BVHLoader.cc.o:(.data.rel.ro+0x0): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/SystemPaths.cc.o:(.rodata+0x0): multiple definition of `typeinfo name for SingletonT<gazebo::common::ModelDatabase>'; CMakeFiles/gazebo_common.dir/ModelDatabase.cc.o:(.rodata+0x30): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/SystemPaths.cc.o:(.data.rel.ro+0x0): multiple definition of `typeinfo for SingletonT<gazebo::common::ModelDatabase>'; CMakeFiles/gazebo_common.dir/ModelDatabase.cc.o:(.data.rel.ro+0x8): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/SystemPaths.cc.o:(.rodata+0x30): multiple definition of `typeinfo name for SingletonT<gazebo::common::SystemPaths>'; CMakeFiles/gazebo_common.dir/BVHLoader.cc.o:(.rodata+0x0): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/SystemPaths.cc.o:(.data.rel.ro+0x8): multiple definition of `typeinfo for SingletonT<gazebo::common::SystemPaths>'; CMakeFiles/gazebo_common.dir/BVHLoader.cc.o:(.data.rel.ro+0x0): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/MeshCSG.cc.o:(.rodata+0x1c): multiple definition of `typeinfo name for SingletonT<gazebo::common::MeshManager>'; CMakeFiles/gazebo_common.dir/MeshManager.cc.o:(.rodata+0x1fc): first defined here
/usr/bin/ld: CMakeFiles/gazebo_common.dir/MeshCSG.cc.o:(.data.rel.ro+0x8): multiple definition of `typeinfo for SingletonT<gazebo::common::MeshManager>'; CMakeFiles/gazebo_common.dir/MeshManager.cc.o:(.data.rel.ro+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [gazebo/common/CMakeFiles/gazebo_common.dir/build.make:805: gazebo/common/libgazebo_common.so.9.19.0] Error 1
make[1]: *** [CMakeFiles/Makefile2:3161: gazebo/common/CMakeFiles/gazebo_common.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
`

dongpan90, 2022-04-14:
I see this problem maybe cause by "an explicit instantiation definition shall appear at most once in a program"  ,used the compiler for arm:
pi@pi-desktop:~/Desktop$ g++ --version
g++ (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.

j-rivero, 2022-08-03:
I'm seeing the same failure for Gazebo 11.10.2 armhf https://launchpadlibrarian.net/616201206/buildlog_ubuntu-jammy-armhf.gazebo_11.10.2+dfsg-1osrf2_BUILDING.txt.gz

samuk, 2022-10-04:
Afaik this os the bug stopping the packages being built https://github.com/osrf/gazebo/issues/3208 It's unclear to me if anyone is working on it.

PR #3283 (2023-10-10): Remove GZ_SINGLETON_DECLARE definition workaround (fix for armhf)

# 🦟 Bug fix

Fixes #3208. 

## Summary
The trick of defining the template function specialized was introduced [in a commit part of the Windows works](https://github.com/gazebosim/gazebo-classic/commit/7f465242ae537f156353eb59af113b7b786de481#diff-9e0097c29a87bb223bef9013319db8dc6c73d9e559bfc41e3067495dccf8936e) as part of the Windows works, seems to me like a workaround to make the compilation to work somehow. On my Jammy testing system, both clang and gcc compiled it fine with this PR and packages were made using gcc for arm64 [... cut]
closed 2023-10-10 by j-rivero