gazebosim/gazebo-classic #3209: Analysis of external mesh loading libraries and comparing with the existing approach

onurtore, 2022-04-14:
Gazebo's mesh loading library doesn't support glTF2 and FBX mesh loaders. There are two ways to extend the capabilities of the Gazebo's mesh loading system, (A)  Using an external open-source library or (B)  Writing glTF2 and FBX mesh loaders consistent with the current approach.

An initial analysis is required to decide which approach is better. I am planning to generate such analysis in the next couple of days, so any idea is welcome. The main aspect of the analysis will be around: 

1. Maturity analysis.
    Which file types are supported by the external library, development history of the external library.
2. Complexity analysis.
    Complexity comparison of possible external libraries and developing new glTF2 and FBX mesh loaders similar to existing approach.
3. Integration difficulty analysis.
    Integration difficulty comparison of the possible external libraries.    
4. Testing difficulty.
    Testing difficulty comparison of the possible external libraries versus the existing approach. 
5. Required time to Implement vs Integrate Analysis

onurtore, 2022-04-14:
I created an additional document for this analysis in [here](https://docs.google.com/document/d/1bs38PMkIK1B6EcN0l4knN2lFnWhiU4hVGH9SBK0zUMU/edit?usp=sharing). It is empty now, I will start my analysis with  assimp

onurtore, 2022-04-15:
Is it possible to get any feedback of this approach @ahcorde ?

iche033, 2022-04-15:
hi, thanks for looking into this. Since the doc mentions GSOC, we'll be targeting this functionality to the [Ignition / Gazebo](https://ignitionrobotics.org/home) simulator. The mesh loaders have moved to one its libraries here: https://github.com/ignitionrobotics/ign-common/tree/ign-common4/graphics/src

I think you're on the right direction with your analysis. `assimp` is promising if it has good support for the 2 formats already. If we're to writer the loader ourselves, I think glTF2 importer is easier to write as its a open format but there may be more work with fbx as there is a text and also a binary format.

onurtore, 2022-04-15:
Thanks for valuable feedback, I will update my documentations.

onurtore, 2022-04-17:
This topic moved to here: https://github.com/ignitionrobotics/ign-common/issues/344