gazebosim/gazebo-classic #3230: support plowing effect in 2D

deepanshubansal01, 2022-06-16:
It would be nice to support plowing effect in 2D, further details on plowing effect in 1D can be found here https://github.com/osrf/gazebo/issues/3085, https://github.com/osrf/gazebo/pull/3164

scpeters, 2022-06-17:
Are you referring to a lateral plowing effect? #3164 only includes a longitudinal plowing effect

deepanshubansal01, 2022-06-17:
> Are you referring to a lateral plowing effect? #3164 only includes a longitudinal plowing effect

yes, lateral only sorry for the confusion. I have edited the description.

scpeters, 2022-07-06:
I think the need for this feature is most obvious when considering the plowing_effect_test world that uses spheres from #3229.

One approach would be use emulate the behavior of the friction cone model and use the wheel linear velocity unit vector as the longitudinal unit vector in the plowing angle calculations instead of deriving it from `fdir1`. This could be an optional parameter.