gazebosim/gazebo-classic #3236: Apt package for Arm64 Ubuntu 22.04

Aposhian, 2022-07-08:
Currently, there is an apt package published for gazebo on amd64 for Ubuntu 22.04, but not for arm64. Is there a technical reason for this, or is it just a task to cross-compile and release it?

EzraBrooks, 2022-07-22:
As someone who primarily runs my simulations inside Ubuntu VMs on an Apple Silicon Mac, this is blocking me from upgrading to Jammy 😞

EzraBrooks, 2022-07-27:
Hi @chapulina, @vatanaksoytezer pointed me in your direction as someone to ping about the status of this. Is there anything I can do to help get this done? I and a couple other folks at PickNik are using Apple Silicon Macs these days.

j-rivero, 2022-08-03:
> Currently, there is an apt package published for gazebo on amd64 for Ubuntu 22.04, but not for arm64. Is there a technical reason for this, or is it just a task to cross-compile and release it?

We are not releasing Gazebo11 binary packages inside Open Robotics infrastructure anymore starting with Ubuntu 22.04. Instead of that we rely on community efforts (we try to help in any of them) for packaging the software. In the case of Ubuntu/Debian we have been contributing to Debian to maintain the Gazebo package there as an official Debian package. Ubuntu usually sync the packages from Debian when a new release is under development.

For Ubuntu gazebo11 official packages they are only available for amd64. The reason is that the package inside Debian/Ubuntu depends on `libqtpropertybrowser-dev` (which is amd64 only). When we released the package inside our own infrastructure at Open Robotics we embedded part of the code of  `libqtpropertybrowser-dev` so the platform limitation did not appear.

To help you in this case, I've setup a Launchpad PPA to release Gazebo11 using the embedded code, you can find the packages for arm64 and ppc64el at https://launchpad.net/~openrobotics/+archive/ubuntu/gazebo11-non-amd64. 

[armhf build](https://launchpadlibrarian.net/616201206/buildlog_ubuntu-jammy-armhf.gazebo_11.10.2+dfsg-1osrf2_BUILDING.txt.gz) seems to be affected by https://github.com/osrf/gazebo/issues/3208. Should not be difficult to fix, PRs are welcome in the case that anyone has interest on having it.

EzraBrooks, 2022-08-03:
thanks @j-rivero. my application appears to still be failing to build after adding the PPA, due to missing the `ros-humble-gazebo-ros-pkgs` package. Is it feasible to get that in the PPA as well, since it's not in the OSRF or Ubuntu repos?

j-rivero, 2022-08-03:
> thanks @j-rivero. my application appears to still be failing to build after adding the PPA, due to missing the `ros-humble-gazebo-ros-pkgs` package. Is it feasible to get that in the PPA as well, since it's not in the OSRF or Ubuntu repos?

ROS packages are a bit different since they are typically hosted in `packages.ros.org` and being built in the ROS buildfarm. I can propose the and handle the inclusion in ROS repository so packages for Humble and others are built for arm64 in the same way than for amd64. Before doing I would like to check and fix a couple of small things:
  * Verification that the PPA packages actually work on arm64 (Gazebo GUI appears and is usable)
  * Fix for the armhf build problem #3208

EzraBrooks, 2022-08-03:
On my Jammy VM with the PPA and `packages.ros.org` both included in the sources, I get:

```The following packages have unmet dependencies:
 gazebo : Depends: gazebo-common (= 11.10.2+dfsg-1osrf2) but it is not going to be installed
          Recommends: gazebo-plugin-base but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

j-rivero, 2022-08-05:
> On my Jammy VM with the PPA and `packages.ros.org` both included in the sources, I get:

Umm, may I get a Dockerfile that reproduces the problem, please?

EzraBrooks, 2022-08-05:
here's a quick and dirty example based on `ros:humble` that's exhibiting the same issue as my VM.

```Dockerfile
FROM --platform=arm64 ros:humble

RUN apt update && apt install -yyq software-properties-common
RUN add-apt-repository ppa:openrobotics/gazebo11-non-amd64
RUN apt update
RUN apt-get install gazebo
```

output:
```
 > [5/5] RUN apt-get install gazebo:
#8 0.173 Reading package lists...
#8 0.517 Building dependency tree...
#8 0.594 Reading state information...
#8 0.630 Some packages could not be installed. This may mean that you have
#8 0.630 requested an impossible situation or if you are using the unstable
#8 0.630 distribution that some required packages have not yet been created
#8 0.630 or been moved out of Incoming.
#8 0.630 The following information may help to resolve the situation:
#8 0.630
#8 0.630 The following packages have unmet dependencies:
#8 0.675  gazebo : Depends: gazebo-common (= 11.10.2+dfsg-1osrf2) but it is not going to be installed
#8 0.675           Recommends: gazebo-plugin-base but it is not going to be installed
#8 0.679 E: Unable to correct problems, you have held broken packages.
```

samuk, 2022-08-05:
I was trying to solve this and found: http://packages.osrfoundation.org/gazebo/ubuntu-nightly/dists/jammy/main/binary-arm64/

So I did

`sudo nano /etc/apt/sources.list`

Paste: deb http://packages.osrfoundation.org/gazebo/ubuntu-nightly jammy main

```
wget  http://packages.osrfoundation.org/gazebo.key
sudo apt-key add gazebo.key
sudo apt-get update
sudo apt-cache search gazebo
```

gazebo-common - Open Source Robotics Simulator - Shared files
gazebo-doc - Open Source Robotics Simulator - Documentation
libignition-gazebo-dev - Ignition Gazebo classes and functions for robot apps - Development files
libignition-gazebo-plugins - Ignition Gazebo classes and functions for robot apps - Plugins
libignition-gazebo6-6 - Ignition Gazebo classes and functions for robot apps - Shared library
ros-simulators-dev - Robot OS simulators development metapackage
libignition-gazebo7 - transitional package
libignition-gazebo7-dbg - transitional package
libignition-gazebo7-dev - transitional package
libignition-gazebo7-plugins - transitional package
python3-ignition-gazebo7 - transitional package
ros-humble-gazebo-dev - Provides a cmake config for the default version of Gazebo for the ROS distribution.
ros-humble-gazebo-msgs - Message and service data structures for interacting with Gazebo from ROS2.
ros-humble-gazebo-msgs-dbgsym - debug symbols for ros-humble-gazebo-msgs
ros-humble-menge-vendor - Menge is a powerful, cross-platform, modular framework for crowd simulation developed at the University of North Carolina - Chapel Hill.
ros-humble-rmf-building-sim-common - Common utility functions for Gazebo and ignition building plugins
ros-humble-rmf-building-sim-ignition-plugins - Ignition gazebo plugins for building infrastructure simulation
ros-humble-rmf-robot-sim-common - Common utility functions for Gazebo and ignition RMF plugins

I was just about to try and install, am I better off using the PPA instead?

samuk, 2022-08-05:
Same here (with the deb http://packages.osrfoundation.org/gazebo/ubuntu-nightly jammy main) and PPA enabled

sudo su
root@ubuntu:/home/ubuntu/linorobot2_ws# apt update && apt install -yyq software-properties-common
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease                                                                 
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease                                                               
Hit:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease                             
Hit:5 https://apt.foxglove.dev/studio stable InRelease                                          
Hit:6 http://packages.ros.org/ros2/ubuntu jammy InRelease           
Get:7 http://packages.osrfoundation.org/gazebo/ubuntu-nightly jammy InRelease [4,283 B]
Fetched 4,283 B in 3s (1,288 B/s)    
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: http://packages.osrfoundation.org/gazebo/ubuntu-nightly/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists...
Building dependency tree...
Reading state information...
software-properties-common is already the newest version (0.99.22.2).
software-properties-common set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@ubuntu:/home/ubuntu/linorobot2_ws# add-apt-repository ppa:openrobotics/gazebo11-non-amd64
PPA publishes dbgsym, you may need to include 'main/debug' component
Repository: 'deb https://ppa.launchpadcontent.net/openrobotics/gazebo11-non-amd64/ubuntu/ jammy main'
Description:
The PPA hosts the Gazebo11 non amd64 versions of the simulator for the Ubuntu versions that have not them and also not present in packages.osrfoundation.org (support ended in Focal).
More info: https://launchpad.net/~openrobotics/+archive/ubuntu/gazebo11-non-amd64
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
Adding deb entry to /etc/apt/sources.list.d/openrobotics-ubuntu-gazebo11-non-amd64-jammy.list
Adding disabled deb-src entry to /etc/apt/sources.list.d/openrobotics-ubuntu-gazebo11-non-amd64-jammy.list
Adding key to /etc/apt/trusted.gpg.d/openrobotics-ubuntu-gazebo11-non-amd64.gpg with fingerprint BD832F46DA885E3353629A832FE9B048E92252B9
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease                                                                                       
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease                                                                                     
Hit:4 https://apt.foxglove.dev/studio stable InRelease                                                                       
Hit:5 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease                                                          
Get:6 https://ppa.launchpadcontent.net/openrobotics/gazebo11-non-amd64/ubuntu jammy InRelease [24.4 kB]                      
Hit:7 http://packages.ros.org/ros2/ubuntu jammy InRelease                  
Get:8 http://packages.osrfoundation.org/gazebo/ubuntu-nightly jammy InRelease [4,283 B]
Get:9 https://ppa.launchpadcontent.net/openrobotics/gazebo11-non-amd64/ubuntu jammy/main arm64 Packages [1,820 B]
Get:10 https://ppa.launchpadcontent.net/openrobotics/gazebo11-non-amd64/ubuntu jammy/main Translation-en [796 B]
Fetched 31.3 kB in 4s (8,550 B/s)                           
Reading package lists... Done
W: http://packages.osrfoundation.org/gazebo/ubuntu-nightly/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
root@ubuntu:/home/ubuntu/linorobot2_ws# apt update
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease                                                                                       
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease                                                                                     
Hit:4 https://apt.foxglove.dev/studio stable InRelease                                                                                                   
Hit:5 https://ppa.launchpadcontent.net/openrobotics/gazebo11-non-amd64/ubuntu jammy InRelease
Hit:6 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease                       
Hit:7 http://packages.ros.org/ros2/ubuntu jammy InRelease           
Get:8 http://packages.osrfoundation.org/gazebo/ubuntu-nightly jammy InRelease [4,283 B]
Fetched 4,283 B in 4s (1,173 B/s)   
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: http://packages.osrfoundation.org/gazebo/ubuntu-nightly/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
root@ubuntu:/home/ubuntu/linorobot2_ws# apt-get install gazebo
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo : Depends: gazebo-common (= 11.10.2+dfsg-1osrf2) but it is not going to be installed
          Recommends: gazebo-plugin-base but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
root@ubuntu:/home/ubuntu/linorobot2_ws#

j-rivero, 2022-08-08:
Umm interesting. There was a conflict with a non-platform package (gazebo-common) and the require of the PPA exact version. I've relaxed the restriction since gazebo-common package should be compatible and almost the same (if not identical). Could you please update and give a try to version `11.10.2+dfsg-1osrf3`?

EzraBrooks, 2022-08-08:
After that change, it does install correctly and I am able to launch the Gazebo GUI and add objects to the simulation 🙂

samuk, 2022-08-08:
Yes, nice works for me too. Thanks

j-rivero, 2022-08-08:
> After that change, it does install correctly and I am able to launch the Gazebo GUI and add objects to the simulation slightly_smiling_face

Thanks guys for testing. Happy to heard that the simulator is working for you.

> The script I'm trying to run still fails as ros-humble-gazebo-ros-pkgs doesn't yet exist.

Yep, for that to happen we need a few more steps. Mainly:
 * Patch the code to compile on armhf see [armhf build](https://launchpadlibrarian.net/616201206/buildlog_ubuntu-jammy-armhf.gazebo_11.10.2+dfsg-1osrf2_BUILDING.txt.gz) seems to be affected by https://github.com/osrf/gazebo/issues/3208.
 * As soon as I get the patch for that I'll build the `armhf` version and import the packages to the ROS repo.
 * At that point, the buildfarm should produce the `ros-humble-gazebo-ros-pkgs` binaries from now on.

EzraBrooks, 2022-08-09:
Is `armhf` a supported platform for ROS? `arm64` Ubuntu is a tier 1 platform according to OSRF. I don't have an `armhf` machine on hand to fix this issue, otherwise I would be happy to contribute a fix.

I have some coworkers on `arm64` laptops to whom I'm handing a codebase off to next week that still uses gazebo11, so it would be nice to have the ROS packages on hand, even if we need to grab them from the nightlies - do we need to block on this `armhf` issue?

samuk, 2022-08-11:
I only find one reference to [BVHlLoader.cc ](https://github.com/osrf/gazebo/blob/7a43a8d23b61cda57ab587b7f7032d9887ba9931/gazebo/common/CMakeLists.txt#L29)in the code

j-rivero, 2022-08-12:
For a testing `armhf` platform, we can probably use a Dockerfile:
```
FROM --platform=armhf ubuntu:jammy

RUN apt update && apt install -yyq software-properties-common build-essential
RUN add-apt-repository -s ppa:openrobotics/gazebo11-non-amd64
RUN apt update
RUN apt-get source gazebo
```

and `docker buildx build .` + `docker run`.

> Is `armhf` a supported platform for ROS? `arm64` Ubuntu is a tier 1 platform according to OSRF. I don't have an `armhf` machine on hand to fix this issue, otherwise I would be happy to contribute a fix.

While `armhf` is not a supported platform in tier 1 for ROS we have typically request that platform for users of Gazebo standalone and I would like to have too if possible. Thanks!

EzraBrooks, 2022-08-12:
I spent some time with the above solution (`armhf` emulation through Docker) between my last comment and now, but unfortunately I'm not skilled enough in CMake to diagnose the issue. My initial instinct was that it had something to do with position-independent code, since I've seen that before, but after some simple debugging and dumping the compile commands, it still wasn't working even with `-fPIC` on, as far as I could tell, every library. But take this with a grain of salt - I am not a C++ programmer by trade 😞

samuk, 2022-09-23:
Any updates on this? Would be great to get Gazebo on Humble/Arm

KilianFt, 2022-09-24:
I made it work with installing gazebo as mentioned above, instead of
```
sudo apt-get install gazebo
```
I used
```
sudo apt-get install libgazebo-dev
```
as I needed the additional packages.   
Regarding ros-humble-gazebo-ros-pkgs, just download it from their github and build it locally like any other ROS package with colcon build. Worked both in docker and native on Ubuntu 22.04 with arm64.

EzraBrooks, 2022-09-26:
I've been building `ros-humble-gazebo-ros-pkgs` from source as well, but it has introduced some library version conflicts with some other packages in my project, which is annoying.

pgarcia-dev, 2022-10-04:
Hi all, 
First of all thank you for this post 🙂

I am facing the same problem described here under arm64. I was able to install gazebo and make it work with the given explanation above but I can’t install `ros-humble-gazebo-ros-pkgs` neither `ros-rolling-gazebo-ros-pkgs`  

I see these packages do exist for Ubuntu focal:
http://packages.ros.org/ros2/ubuntu/dists/focal/main/binary-arm64/Packages
but they don’t for Ubuntu jammy:
http://packages.ros.org/ros2/ubuntu/dists/jammy/main/binary-arm64/Packages

> I've been building `ros-humble-gazebo-ros-pkgs` from source as well, but it has introduced some library version conflicts with some other packages in my project, which is annoying.

Same here

is it planned to add all gazebo related to Ros packages to the OSRF jammy repo for humble and/or rolling?

did anyone find a workaround?

samuk, 2022-10-04:
Afaik this os the bug stopping the packages being built https://github.com/osrf/gazebo/issues/3208 It's unclear to me if anyone is working on it.

pgarcia-dev, 2022-10-05:
Does anyone know if anybody is working and that? @j-rivero ?

j-rivero, 2022-10-05:
> Does anyone know if anybody is working and that? @j-rivero ?

Should not be difficult to fix but I did not have time to get into it. Count me in for reviewing the patch and doing the release and import to the ROS buildfarm but I would appreciate if someone can please send the patch for #3208.

j-rivero, 2022-11-29:
> 
> Should not be difficult to fix but I did not have time to get into it. Count me in for reviewing the patch and doing the release and import to the ROS buildfarm but I would appreciate if someone can please send the patch for #3208.

I made the fix for the arm version and send the patch to the code: https://github.com/gazebosim/gazebo-classic/pull/3283 and build the packages for armhf, arm64 and ppc64 in https://launchpad.net/~openrobotics/+archive/ubuntu/gazebo11-non-amd64.

Unfortunately the packages are not going to be imported to the ROS repository for some good reasons, see https://github.com/osrf/gazebo_tutorials/pull/169/. From here you have a couple of options: build `gazebo_ros_pkgs` from source using the PPA binaries for Gazebo11 on arm64 or migrate to the new Gazebo if that is an option so you get full support and binaries.

stephan-jacobs, 2023-08-05:
Hi there, I'm trying to get gazebo11 working on ARM for a university course, but I'm not experienced with ROS yet. 

If I want to use the packages in the PPA provided by @j-rivero , what should I do exactly? 

(general tips are fine I just don't know what instructions to look for at this stage)
Thanks for your time!

AndromedaHelix, 2023-12-18:
Ive installed the PPA from https://launchpad.net/~openrobotics/+archive/ubuntu/gazebo11-non-amd64, but how do I open gazebo?

Also, about building from source, how do I do that? Im new to ros and dont know how to do that.

AndromedaHelix, 2023-12-20:
While downloading gazebo from https://launchpad.net/~openrobotics/+archive/ubuntu/gazebo11-non-amd64, it did install, however I can't place any models and it freezes almost every time I open the app. I don't know if I did anything wrong @j-rivero ?

AndromedaHelix, 2023-12-25:
I get this issue when running `gazebo --verbose` 
`[Err] [TopicManager.cc:321] Shouldn't get here topic[/gazebo/default/request]`

And then gazebo freezes. I can't find any solutions to the issue.

SebastianBitsch, 2024-03-06:
Is there currently a working solution for Gazebo with Arm and Ubuntu 22.04? The workarounds mentioned above don't seem to work, ```libgazebo-dev``` seems to [be removed](https://www.ubuntuupdates.org/package/core/jammy/universe/proposed/libgazebo-dev) and the version '11.10.2+dfsg-1osrf3' wasn't found. 
Any minimal working docker file would be of great help, thanks

closed 2024-03-08 by j-rivero

romain130492, 2024-08-24:
hell, any news about this?
What are the steps to follow to install gazebo on an ARM machine running for example: ubuntu 22 on parallel desktop?

With ros2Humble

romain130492, 2024-08-24:
> Unfortunately the packages are not going to be imported to the ROS repository for some good reasons, see [osrf/gazebo_tutorials#169](https://github.com/osrf/gazebo_tutorials/pull/169). From here you have a couple of options: build `gazebo_ros_pkgs` from source using the PPA binaries for Gazebo11 on arm64 or migrate to the new Gazebo if that is an option so you get full support and binaries.

Can i have some clarification on the steps to achieve that? thanks

garden666, 2024-10-10:
I have installed Gazebo on Ubuntu under the aarch framework. However, when running a program, I found that gazebo_ros is missing. I ran "sudo apt install ros-humble-gazebo-ros-pkgs" but to no avail. It shows that this package cannot be located. Then I added the ppa source as mentioned above and tried again, but still without success. Now I don't know what to do.