gazebosim/gazebo-classic #3242: Runtime errors on macOS with protobuf 21.2

scpeters, 2022-07-21:
Protobuf version 21.2 was recently merged in homebrew-core, and since that time there have been run-time errors in all gazebo tests that link against protobuf with lazy symbol binding errors like the following:

~~~
1: [ RUN      ] MsgsTest.Msg
1: dyld: lazy symbol binding failed: Symbol not found: __ZN6google8protobuf8internal16InternalMetadataD1Ev
1:   Referenced from: /Users/jenkins/workspace/gazebo-ci-pr_any-homebrew-amd64/build/gazebo/msgs/libgazebo_msgs.9.dylib
1:   Expected in: flat namespace
1: 
1: dyld: Symbol not found: __ZN6google8protobuf8internal16InternalMetadataD1Ev
1:   Referenced from: /Users/jenkins/workspace/gazebo-ci-pr_any-homebrew-amd64/build/gazebo/msgs/libgazebo_msgs.9.dylib
1:   Expected in: flat namespace
1: 
  1/662 Test   #1: UNIT_msgs_TEST ........................................Subprocess aborted***Exception:   0.33 sec
~~~

With `c++filt` the symbol `__ZN6google8protobuf8internal16InternalMetadataD1Ev` is translated to `google::protobuf::internal::InternalMetadata::~InternalMetadata()`. It turns out there is an ABI with that symbol reported in https://github.com/protocolbuffers/protobuf/issues/9947 that has been fixed in the next version of protobuf (21.3). This should be fixed once the next version of protobuf is merged to homebrew-core from https://github.com/Homebrew/homebrew-core/pull/106252.

carlocab, 2022-07-21:
You can also pass `-DNEDEBUG` to your builds so you don't need to wait for the new version. We've applied this fix to a number of `protobuf` dependents in Homebrew/core.

scpeters, 2022-07-21:
thanks @carlocab! it looks like you've already merged 21.3 with the fix, and now my bottles are working again!

closed 2022-07-21 by scpeters