gazebosim/gazebo-classic #3266: It is not possible to installing Gazebo on fresh Ubuntu

Aladdin-Ark, 2022-10-08:
It seems that using the official steps to install gazebo is not working with the latest updates ignition common3

```
Preparing to unpack .../libignition-common3-av-dev_3.15.0-1~focal_amd64.deb ...
Unpacking libignition-common3-av-dev:amd64 (3.15.0-1~focal) ...
dpkg: error processing archive /var/cache/apt/archives/libignition-common3-av-dev_3.15.0-1~focal_amd64.deb (--unpack):
 trying to overwrite '/usr/include/ignition/common3/gz/common/av.hh', which is also in package libignition-common3-core-dev:amd64 3.15.0-1~focal

Preparing to unpack .../libignition-common3-events-dev_3.15.0-1~focal_amd64.deb ...
Unpacking libignition-common3-events-dev:amd64 (3.15.0-1~focal) ...
dpkg: error processing archive /var/cache/apt/archives/libignition-common3-events-dev_3.15.0-1~focal_amd64.deb (--unpack):
 trying to overwrite '/usr/include/ignition/common3/gz/common/events.hh', which is also in package libignition-common3-core-dev:amd64 3.15.0-1~focal

Preparing to unpack .../libignition-common3-graphics-dev_3.15.0-1~focal_amd64.deb ...
Unpacking libignition-common3-graphics-dev:amd64 (3.15.0-1~focal) ...
dpkg: error processing archive /var/cache/apt/archives/libignition-common3-graphics-dev_3.15.0-1~focal_amd64.deb (--unpack):
 trying to overwrite '/usr/include/ignition/common3/gz/common/graphics.hh', which is also in package libignition-common3-core-dev:amd64 3.15.0-1~focal
```

MarcelDudek, 2022-10-08:
Yes, I had the same problem when upgrading with `apt`. Had to manually downgrade and hold all `libignition-common3` packages at version 3.14.2-1.

So if you want to install Gazebo on Ubuntu at the moment, one workaround is to manually install `libignition-common3` packages with
```bash
sudo apt install "libignition-common3*=3.14.2-1~focal"
```
before Gazebo installation.

Aladdin-Ark, 2022-10-09:
I was trying to do the same as you suggested, but using `apt-cache policy libignition-common3-dev` showing no other candidates than 3.15.0-1 on Ubuntu focal

MarcelDudek, 2022-10-09:
> I was trying to do the same as you suggested, but using `apt-cache policy libignition-common3-dev` showing no other candidates than 3.15.0-1 on Ubuntu focal

Yeah, candidate for 3.14 is available in ROS2 repository: http://packages.ros.org/ros2/ubuntu, so you can try to add it to your sources.

mikaelarguedas, 2022-10-10:
Confirmed that this started to fail on CI 3 days ago. It seems to come from the 3.15.0 release.
cc @j-rivero to be sure you're aware of this breakage :)

MatthijsBurgh, 2022-10-10:
This is not related to Gazebo, but the ignition libs. So not really the correct title and place for this issue. But I experience the same issue.

Aladdin-Ark, 2022-10-10:
I am not sure what issue you are experiencing when you say it is not related to Gazebo 
But let me explain the details to be clear

Using the official steps to install Gazebo on ubuntu It is not working because of an issue in the provided packages of `libignition-common3` with version `3.15.0-1` 
I understand I can downgrade and install `3.14.2-1` but it is not available on the `osrf/gazebo` repo anymore so it is still not possible to install gazebo on a fresh ubuntu following the official steps.

I think from here you can redirect to the right place rather....

update: 
`libignition-common3` have released `3.15.0-3` versions and seems the problem is fixed!

closed 2022-10-10 by Aladdin-Ark