gazebosim/gazebo-classic #3277: Gazebo Classic - cannot install in Ubuntu 22.04 + ROS2 Humble, error of unmet dependencies (libignition-common3-dev, libgazebo11, ...)

ljaniec, 2022-11-15:
## Environment
* OS Version: Ubuntu 22.04
* Binary build, Gazebo Classic 11
* ROS2 Humble

## Description
* Expected behavior: I can install `sudo apt install gazebo` and `sudo apt install ros-humble-gazebo-ros-pkgs` to use Gazebo Classic with [the Nav2 tutorial](https://navigation.ros.org/getting_started/index.html)
* Actual behavior: Installation from binaries doesn't work, I tried building from source as [here](https://classic.gazebosim.org/tutorials?tut=install_from_source&cat=install), but I also got errors of unmet dependencies:
 `libignition-common3-dev : Depends: libignition-common3-core-dev` etc.

## Steps to reproduce

I upgraded Ubuntu 20.04 to the 22.04 version. I installed ROS2 Humble (`sudo apt install ros-humble-desktop-full`) and I wanted to use Gazebo Classic with Nav2 simulations, but I cannot use it - my packages have unmet dependencies, that I couldn't fix. 
Please hint me what I should do now, it seems like a bug.

My terminal logs:

- When I tried normal `sudo apt install gazebo`:

>     ljaniec@ljaniec-PC:~$ gazebo
> 
>     Command 'gazebo' not found, but can be installed with:
> 
>     sudo apt install gazebo
> 
>     ljaniec@ljaniec-PC:~$ sudo apt install gazebo
> 
>     Reading package lists... Done
> 
>     Building dependency tree... Done
> 
>     Reading state information... Done
> 
>     Some packages could not be installed. This may mean that you have
> 
>     requested an impossible situation or if you are using the unstable
> 
>     distribution that some required packages have not yet been created
> 
>     or been moved out of Incoming.
> 
>     The following information may help to resolve the situation:
> 
>     
>     The following packages have unmet dependencies:
> 
>      gazebo : Depends: libgazebo11 (= 11.10.2+dfsg-1) but 11.12.0-1~focal is to be installed
> 
>     E: Unable to correct problems, you have held broken packages.

- When I tried `sudo apt install ros-humble-gazebo-ros-pkgs`:

>     ljaniec@ljaniec-PC:~$ sudo apt install ros-humble-gazebo-ros-pkgs 
> 
>     Reading package lists... Done
> 
>     Building dependency tree... Done
> 
>     Reading state information... Done
> 
>     Some packages could not be installed. This may mean that you have
> 
>     requested an impossible situation or if you are using the unstable
> 
>     distribution that some required packages have not yet been created
> 
>     or been moved out of Incoming.
> 
>     The following information may help to resolve the situation:
>     
>     The following packages have unmet dependencies:
>     
>      gazebo : Depends: libgazebo11 (= 11.10.2+dfsg-1) but 11.12.0-1~focal is to be installed
>     
>      gz-tools2 : Conflicts: gazebo (>= 11.0.0) but 11.10.2+dfsg-1 is to be installed
>     
>      libgazebo-dev : Depends: libgazebo11 (= 11.10.2+dfsg-1) but 11.12.0-1~focal is to be installed
>     
>     E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
>     
>     ljaniec@ljaniec-PC:~$ sudo apt install libgazebo11
>     
>     Reading package lists... Done
>     
>     Building dependency tree... Done
>     
>     Reading state information... Done
>     
>     libgazebo11 is already the newest version (11.12.0-1~focal).
>     
>     0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

- When I tried `sudo apt install -f ros-humble-gazebo-ros-pkgs`
> ljaniec@ljaniec-PC:~/ros2_ws$ sudo apt install -f ros-humble-gazebo-ros-pkgs
[sudo] password for ljaniec: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
>
> The following packages have unmet dependencies:
 libignition-common3-dev : Depends: libignition-common3-core-dev (= 3.14.0+dfsg-5) but 3.15.1-1\~focal is to be installed
                           Depends: libignition-common3-av-dev (= 3.14.0+dfsg-5) but 3.15.1-1~focal is to be installed
E: Unable to correct problems, you have held broken packages.

j-rivero, 2022-11-15:
Thansk @ljaniec for the detailed bug. I was able to reproduce the problem. Here is what I think it is happening:

 * You had an Ubuntu 20.04 system using `packages.osrfoundation.org` repository to install the latest Gazebo11, `11.12.0-1~focal`  at your system. 
 * You had upgraded the system from 20.04 to 22.04 while keeping the Gazebo11 packages installed so you end up with some focal packages being installed in the 22.04 system. This is expected.
 * Ubuntu 20.04 (Focal) is the latest Ubuntu supported by binary packages for Gazebo11 from `packages.osrfoundation.org`. In Ubuntu 22.04 (Jammy) the Gazebo11 version comes directly from Ubuntu repositories and it rarely changes, being `11.10.2+dfsg-1`.
 * When you upgrade, the system found some discrepancies among the packages versions, since the packages in Ubuntu don't expect packages coming from non official sources, the upgrade path is not prepared, the packages manager found a broken situation.

The solution would be to uninstall focal Gazebo + dependencies and reinstall Gazebo (package names changed a bit, `gazebo` and `libgazebo-dev` without the 11). Not that easy with the broken state of the packages database.

A possible solution: the version of the packages coming from `packages.osrfoundation.org` have the suffix `~focal`. We can brute force a little bit the packages system to get rid of the situation by uninstalling ignition, sdformat and Gazebo from `~focal` by using directly `dpkg -r`.

```bash
# Check what is going to be removed
dpkg -l | grep '^ii.*\(libignition\|sdformat\|gazebo\).*~focal'
# Remove everything
while [[ -n $(dpkg -l | grep '^ii.*\(libignition\|sdformat\|gazebo\).*~focal' | awk '{ print $2 };') ]]; do sudo dpkg -r $(dpkg -l | grep '^ii.*\(libignition\|sdformat\|gazebo\).*~focal' | awk '{ print $2 }';); done
# Install new Gazebo on Jammy (or just install the humble desktop full)
sudo apt install libgazebo-dev
```

Let me know if this works for you. Sorry for the inconvenience.

ljaniec, 2022-11-15:
I fixed it already, the brute way of purging packages and `sudo dpkg -i --force-overwrite <file-path>` (e.g. files like `/var/cache/apt/archives/libignition-common3-3_3.14.0+dfsg-5_amd64.deb` etc.) as mentioned [here](https://askubuntu.com/questions/176121/dpkg-error-trying-to-overwrite-file-which-is-also-in) helped, together with deleting disabled repositories in the "Other Software" in the Software & Updates. 

To be honest, I am not sure what was the culprit in the end, I didn't get to check your solution.
If I can get you additional logs or anything else to pinpoint it, feel free to ask :)

closed 2022-11-15 by scpeters

j-rivero, 2022-11-15:
> To be honest, I am not sure what was the culprit in the end, I didn't get to check your solution.

I think that the process of upgrading a system from Focal to Jammy with Gazebo installed is going to leave the gazebo packages broken. So no your fault. Not sure if there is too much we can do since we are not in control of the Gazebo packages in Jammy (official from Ubuntu).