gazebosim/gazebo-classic #3281: Conflict: libignition-transport8-core-dev_8.4.0-2~focal and libignition-transport8-log-dev:amd64 8.4.0-2~focal - Issue upgrading libgazebo11

argenos, 2022-11-21:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build?
  
  

## Description
* Expected behavior:  Updating from the package manager succeeds
* Actual behavior:  Unmet dependencies and package conflicts

## Steps to reproduce

1. Got an error for unmet dependencies while updating libgazebo11
2. Followed steps in https://github.com/gazebosim/gazebo-classic/issues/2995#issuecomment-834221087 to upgrade packages from packages.osrfoundation.org instead
3. Got a conflict while unpacking `libignition-transport8-core-dev_8.4.0-2`

## Output

Unmet dependencies (step 1)

```
The following packages have unmet dependencies.
 libignition-transport8-core-dev : Depends: libignition-transport8 (= 8.3.0-2~focal) but 8.4.0-2~focal is to be installed
 libignition-transport8-dev : Depends: libignition-transport8-core-dev (= 8.4.0-2~focal) but 8.3.0-2~focal is to be installed
 libignition-transport8-log-dev : Depends: libignition-transport8-core-dev (= 8.4.0-2~focal) but 8.3.0-2~focal is to be installed
```

Conflict (from step 3)

```
Preparing to unpack .../libignition-transport8-core-dev_8.4.0-2~focal_amd64.deb ...
Unpacking libignition-transport8-core-dev:amd64 (8.4.0-2~focal) over (8.3.0-2~focal) ...
dpkg: error processing archive /var/cache/apt/archives/libignition-transport8-core-dev_8.4.0-2~focal_amd64.deb (--unpack):
 trying to overwrite '/usr/include/ignition/transport8/gz/transport/log.hh', which is also in package libignition-transport8-log-dev:amd64 8.4.0-2~focal
```

traversaro, 2022-11-21:
Example CI failure: https://github.com/robotology/robotology-superbuild/actions/runs/3513995697/jobs/5887481783 .

Apparently the issue is present both on Ubuntu 18.04 and 20.04 . fyi @j-rivero

traversaro, 2022-11-21:
> Example CI failure: https://github.com/robotology/robotology-superbuild/actions/runs/3513995697/jobs/5887481783 .
> 
> Apparently the issue is present both on Ubuntu 18.04 and 20.04 . fyi @j-rivero

At least on GitHub Actions the problem is not present anymore (see https://github.com/robotology/robotology-superbuild/actions/runs/3517049691/jobs/5894369215). @argenos are you still experiencing the problem?

vahidzolf, 2022-11-22:
We also have this problem when installing those packages from the ubuntu repo. It breaks our platform installation. So, could you please let me know what is the plan for this issue? thanks.

argenos, 2022-11-22:
I can confirm this seems to be solved for me.

traversaro, 2022-11-22:
> I can confirm this seems to be solved for me.

Great! 

> We also have this problem when installing those packages from the ubuntu repo. It breaks our platform installation. So, could you please let me know what is the plan for this issue? thanks.

When did this happened? Did you try `sudo apt update` before upgrading/installing gazebo?

vahidzolf, 2022-11-22:
Yes I was updating apt before getting them. But it works for me now! Thanks.

traversaro, 2022-11-22:
Great, so the problem seems solved now. I think we can close the issue, feel free to re-open/open a new issue if you encounter other problems of this kind.

closed 2022-11-22 by traversaro

argenos, 2022-11-22:
Thanks for the quick answer and fix!

j-rivero, 2022-11-22:
Thanks for the report guys! I saw it first in https://github.com/gazebo-release/gz-transport8-release/issues/10 and hopefully is not going to happen again after merging https://github.com/gazebo-release/gz-transport8-release/pull/8. Sorry for the inconvenience.

traversaro, 2022-11-22:
> Thanks for the report guys! I saw it first in [gazebo-release/gz-transport8-release#10](https://github.com/gazebo-release/gz-transport8-release/issues/10) and hopefully is not going to happen again after merging [gazebo-release/gz-transport8-release#8](https://github.com/gazebo-release/gz-transport8-release/pull/8). Sorry for the inconvenience.

Thanks for the fix!