gazebosim/gazebo-classic #3284: Ubuntu 22.04 arm build crush

Fricodelco, 2022-11-30:
## Environment
* OS Version: 
* source 
branch jrivero/remove_singleton_definitions

## Description
gazebo crashes after start with no error to terminal. 

## Steps to reproduce
build this commit or install ppa ARM build for ubuntu 22.04

## Output

[image]

j-rivero, 2022-11-30:
Thanks for testing @Fricodelco. We are going to need more info to see what's going on. Could you please:
 * Run `gazebo --verbose` and see if it provides some useful info
 * Run `gzserver --verbose` and `gzclient --verbose` in different terminals to see which one is crashing
 * Run `gzserver` or `gzclient` under gdb or other debugging tool to get a backtrace of the problem.

Thanks!

Fricodelco, 2022-12-01:
i got this error with gzserver --verbose
[image]

j-rivero, 2022-12-01:
The code comes from the [TopicManager.cc](https://github.com/gazebosim/gazebo-classic/blob/gazebo11/gazebo/transport/TopicManager.cc#L304-L323) so it could be related to the network configuration. Could you please run `ip addr show` or `ifconfig` to see how is the network configured? Thanks.

Fricodelco, 2022-12-01:
i run ubuntu 22.04 under parallels with m1 mac, ros2 works well with cyclonedds, so i suppose gazebo must work as well...

[image]

nihavend, 2023-01-28:
I have an installation issue (M1-Parallels-Ubuntu 20.04)

AndromedaHelix, 2023-12-21:
Hi, I'm having the same issue. Has there been any works or solutions to this issue?