gazebosim/gazebo-classic #3289: Gazebo Won't Pause or Allow to add model into world

mora200217, 2023-01-18:
## Environment
* OS Version: macOS Monterey 12.6.2 
* Binary build from conda-forge channel (also tried with robostack) . Gazebo 11.12.0 on Robostack Channel 
  
  

## Description
* Expected behavior: Allow to add elements to the gazebo world, and pause the simulation 
* Actual behavior: The elements added appear while translating them in order to be placed, but when clicked to lock position it disappears and after a while crashes. (Also crashes when trying to pause the simulation; crashes after about 20 s) 

## Steps to reproduce

0. Install gazebo from robostack channel 
1. Start gazebo from console
2. Try to add an element to the world or pause the simulation

> IMPORTANT: I'm using mamba with conda-forge as package manager. I've tried using a python 3.7, 3.8, 3.9 and 3.10 Environment. Either with direct init `$ gazebo` or using gazebo_ros package and launch and empty world, the behaviour is still the same. 

## Output

Verbose output gazebo command when crashed
[image]

mora200217, 2023-01-18:
See RoboStack/ros-noetic#323

traversaro, 2023-01-18:
Hi @mora200217, do you have the same problem also with the Gazebo installed via homebrew?

mora200217, 2023-01-18:
> Hi @mora200217, do you have the same problem also with the Gazebo installed via homebrew?

@traversaro yup. Neither brew nor robostack works.

traversaro, 2023-01-20:
> See [RoboStack/ros-noetic#323](https://github.com/RoboStack/ros-noetic/issues/323)

From this it seems you are using the problematic `qt=5.12.9=h2a607e2_5` build of qt. Can you check if the problem remains even if you install:
~~~
mamba install qt=5.12.9=hda022c4_4
~~~
see https://github.com/conda-forge/gazebo-feedstock/issues/119 for more info.

mora200217, 2023-02-08:
> From this it seems you are using the problematic `qt=5.12.9=h2a607e2_5` build of qt. Can you check if the problem remains even if you install:
> ~~~
> mamba install qt=5.12.9=hda022c4_4
> ~~~
> see [...]

Okay! I will give it a shot

simonbogh, 2023-02-15:
Any luck getting this to work? I am in the same situation, same setup, though `macOS Ventura 13.2`. I have to go back to `Gazebo 11.9` but then the python environment cannot be solved in order to install `ros-noetic-desktop`.

traversaro, 2023-02-16:
> Any luck getting this to work? I am in the same situation, same setup, though `macOS Ventura 13.2`. I have to go back to `Gazebo 11.9` but then the python environment cannot be solved in order to install `ros-noetic-desktop`.

Are you sure you have the same setup @simonbogh ? From what I understand from https://github.com/RoboStack/ros-noetic/issues/323, it seems that you were using qt=5.15, not qt=5.12 )in general when reporting an issue with sofware installed with conda/mamba it is useful to always report the `mamba list`). Anyhow, can you check now after https://github.com/conda-forge/gazebo-feedstock/pull/162 has been merged? The issue should be solved for 11.12.0=*_6 builds.

simonbogh, 2023-02-16:
True @traversaro. I will keep the conversation going on the other issue thread instead, as it seems you have identified the problem and a potential solution there.

traversaro, 2023-05-17:
The issue was fixed by https://github.com/gazebosim/gazebo-classic/pull/3269, see https://github.com/RoboStack/ros-noetic/issues/323 .

closed 2023-05-17 by traversaro