gazebosim/gazebo-classic #3292: ABI checker jobs are failing

scpeters, 2023-01-26:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? CI build from source

## Description
* Expected behavior: ABI checker job should compile successfully
* Actual behavior: it fails to compile (for example in https://github.com/gazebosim/gazebo-classic/pull/3290)

## Steps to reproduce

1. Open a pull request with a minor change, which triggers an ABI checker job
2. The job will fail

## Output

From the [Abi checker log](https://build.osrfoundation.org/job/gazebo-abichecker-any_to_any-ubuntu_auto-amd64/ws/abi_checker/logs/gazebo-classic/X/log.txt/*view*/):

~~~
The GCC parameters:
  gcc -fdump-lang-raw -fkeep-inline-functions -c -x c++ -fpermissive -w  -std=c++11 "/tmp/twsUMx5ApM/dump1.h"  -I/usr/local/destination_branch/include/gazebo-11 -I/usr/local/destination_branch/include/gazebo-11/gazebo -I/usr/local/destination_branch/include/gazebo-11/gazebo/ode -I/usr/include/tbb -I/usr/include/simbody -I/usr/include/sdformat-9.10 -I/usr/include/bullet -I/usr/include/OGRE-2.1 -I/usr/include/OGRE -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/ignition/transport8 -I/usr/include/ignition/msgs5 -I/usr/include/ignition/math6 -I/usr/include/ignition/fuel_tools4 -I/usr/include/ignition/common3 -I/usr/include/ignition/cmake2 -I/usr/include/OGRE/Paging

In file included from /usr/local/destination_branch/include/gazebo-11/gazebo/GIMPACT/gim_memory.h:34,
                 from /usr/local/destination_branch/include/gazebo-11/gazebo/GIMPACT/gim_radixsort.h:35,
                 from /usr/local/destination_branch/include/gazebo-11/gazebo/GIMPACT/gim_boxpruning.h:35,
                 from /usr/local/destination_branch/include/gazebo-11/gazebo/GIMPACT/gim_trimesh.h:33,
                 from /usr/local/destination_branch/include/gazebo-11/gazebo/GIMPACT/gimpact.h:59,
                 from /tmp/twsUMx5ApM/dump1.h:6:
/usr/local/destination_branch/include/gazebo-11/gazebo/GIMPACT/gim_math.h:34:10: fatal error: config.h: No such file or directory
   34 | #include "config.h"
      |          ^~~~~~~~~~
compilation terminated.
~~~

scpeters, 2023-01-26:
I guess we are installing the GIMPACT headers but not their config.h file?

scpeters, 2023-03-17:
I think I've found the underlying issue: https://github.com/gazebo-tooling/release-tools/issues/890

traversaro, 2023-05-17:
Recent builds have a green status, see for example: https://build.osrfoundation.org/job/gazebo-abichecker-any_to_any-ubuntu_auto-amd64/1306/ . However, the output is:
~~~
+ abi-compliance-checker -lang C++ -lib gazebo-classic -old pkg.xml -new devel.xml
Preparing, please wait ...
WARNING: version number #1 is not set (use --v1=NUM option)
Using GCC 9 (x86_64-linux-gnu, target: x86_64)
WARNING: May not work properly with GCC 4.8.[0-2], 6.* and higher due to bug #78040 in GCC. Please try other GCC versions with the help of --gcc-path=PATH option or create ABI dumps by ABI Dumper tool instead to avoid using GCC. Test selected GCC version first by -test and -gcc-path options.
cc1: warning: command line option ‘-std=c++11’ is valid for C++/ObjC++ but not for C
Checking header(s) X ...
ERROR: some errors occurred when compiling headers
ERROR: see log for details:
  /home/jenkins/workspace/gazebo-abichecker-any_to_any-ubuntu_auto-amd64/abi_checker/logs/gazebo-classic/X/log.txt

WARNING: version number #2 is not set (use --v2=NUM option)
cc1: warning: command line option ‘-std=c++11’ is valid for C++/ObjC++ but not for C
Checking header(s) Y ...
ERROR: some errors occurred when compiling headers
ERROR: see log for details:
  /home/jenkins/workspace/gazebo-abichecker-any_to_any-ubuntu_auto-amd64/abi_checker/logs/gazebo-classic/Y/log.txt

Comparing ABIs ...
Comparing APIs ...
Creating compatibility report ...
Binary compatibility: 100%
Source compatibility: 100%
Total binary compatibility problems: 0, warnings: 0
Total source compatibility problems: 0, warnings: 0
Report: compat_reports/gazebo-classic/X_to_Y/compat_report.html
~~~

I am not sure if `ERROR: some errors occurred when compiling headers` can be safely ignored. If yes, probably we can close the issue. @scpeters @j-rivero

scpeters, 2023-05-17:
I think we can close the issue since the build is passing again. I've copied the contents of one of the error logs from the jenkins workspace to the following gist in case anyone wants to follow up:

* https://gist.github.com/scpeters/6ef5dce28aab4d876d173a47850e4af9

closed 2023-05-17 by scpeters