gazebosim/gazebo-classic #3300: An error while instaling gazebo on Ubuntu 22.04.

adricom231, 2023-02-20:
Hi im new to linux i tried instaling gazebo but got this error

---
Finished <<< gz-rendering8 [2.34s]
Starting >>> gz-gui8
--- stderr: gz-gui8                                                                     
CMake Warning at /home/adricom/workspace/install/share/cmake/gz-cmake3/cmake3/GzUtils.cmake:231 (find_package):
  By not providing "Findgz-transport12.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gz-transport12", but CMake did not find one.

  Could not find a package configuration file provided by "gz-transport12"
  with any of the following names:

    gz-transport12Config.cmake
    gz-transport12-config.cmake

  Add the installation prefix of "gz-transport12" to CMAKE_PREFIX_PATH or set
  "gz-transport12_DIR" to a directory containing one of the above files.  If
  "gz-transport12" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:65 (gz_find_package)

CMake Warning at /home/adricom/workspace/install/share/cmake/gz-cmake3/cmake3/GzUtils.cmake:231 (find_package):
  By not providing "Findgz-msgs9.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gz-msgs9",
  but CMake did not find one.

  Could not find a package configuration file provided by "gz-msgs9" with any
  of the following names:

    gz-msgs9Config.cmake
    gz-msgs9-config.cmake

  Add the installation prefix of "gz-msgs9" to CMAKE_PREFIX_PATH or set
  "gz-msgs9_DIR" to a directory containing one of the above files.  If
  "gz-msgs9" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:75 (gz_find_package)

CMake Error at /home/adricom/workspace/install/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:73 (message):
  -- BUILD ERRORS: These must be resolved before compiling.
Call Stack (most recent call first):
  CMakeLists.txt:103 (gz_configure_build)

CMake Error at /home/adricom/workspace/install/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:75 (message):
  -- 	Missing dependency [gz-transport12]
Call Stack (most recent call first):
  CMakeLists.txt:103 (gz_configure_build)

CMake Error at /home/adricom/workspace/install/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:75 (message):
  -- 	Missing dependency [gz-msgs9]
Call Stack (most recent call first):
  CMakeLists.txt:103 (gz_configure_build)

CMake Error at /home/adricom/workspace/install/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:77 (message):
  -- END BUILD ERRORS

Call Stack (most recent call first):
  CMakeLists.txt:103 (gz_configure_build)

CMake Error at /home/adricom/workspace/install/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:82 (message):
  Errors encountered in build.  Please see BUILD ERRORS above.
Call Stack (most recent call first):
  CMakeLists.txt:103 (gz_configure_build)

---
Failed   <<< gz-gui8 [0.33s, exited with code 1]
Aborted  <<< sdformat13 [11.2s]   

Summary: 10 packages finished [14.5s]
  1 package failed: gz-gui8
  1 package aborted: sdformat13
  3 packages had stderr output: gz-fuel_tools9 gz-gui8 gz-rendering8
  4 packages not processed

adricom231, 2023-02-20:
Hi @traversaro i was installing Gazebo Harmonic

traversaro, 2023-02-20:
Then it is the "Modern" Gazebo, not the Gazebo Classic one, so I think you can close the issue here, thanks! Furthermore, Gazebo Harmonic is still in development, so I would suggest to stick to a released version of Gazebo (such as Gazebo Garden) to make sure the installation process works reliably.

traversaro, 2023-02-20:
If you have doubts on how to install Gazebo following the documentation, feel free to post a question on https://answers.gazebosim.org/questions/, thanks!

j-rivero, 2023-02-21:
Please follow https://community.gazebosim.org/t/problem-to-build-gazebo-at-ubuntu-22-04-missing-dependency-gz-msgs9/1813 and specially https://github.com/gazebo-tooling/release-tools/issues/878

closed 2023-02-21 by j-rivero