gazebosim/gazebo-classic #3304: 仿真激光雷达时，如果Gazebo使用的是dart引擎，雷达扫描会穿过物体。但是使用ode可以正常使用不会穿过物体 When simulating laser radar, if Gazebo uses a dart engine, the radar scan will pass through the object. But it can be used normally without passing through objects

huqin-RM, 2023-03-10:
[image]
[image]

When simulating laser radar, if Gazebo uses a dart engine, the radar scan will pass through the object. But it can be used normally without passing through objects

traversaro, 2023-03-11:
I am afraid this is a duplicate of https://github.com/gazebosim/gazebo-classic/issues/911 . Did you tried with gpu_ray ?

huqin-RM, 2023-03-15:
> I am afraid this is a duplicate of #911 . Did you tried with gpu_ray ?

My computer doesn't seem to find gpu ray. How it should be used

traversaro, 2023-03-15:
> using the ode engine can work normally without passing through objects

Then I think it is a duplicate of https://github.com/gazebosim/gazebo-classic/issues/911 .

> > I am afraid this is a duplicate of #911 . Did you tried with gpu_ray ?
> 
> My computer doesn't seem to find gpu ray. How it should be used

See https://github.com/gazebosim/gazebo-classic/blob/d3584008b47c745c1732a6cd07af68726cd900fa/test/worlds/gpu_laser2.world#L43 for an example. 

Given that this is a duplicate issue, I think we can close the issue, what do you think?

huqin-RM, 2023-03-15:
> Then I think it is a duplicate of #911 .
>
>
> See
>
> https://github.com/gazebosim/gazebo-classic/blob/d3584008b47c745c1732a6cd07af68726cd900fa/test/worlds/gpu_laser2.world#L43
>
> for an example.
> Given that [...]

Does Dart engine not support the use of ray related plug-ins at present

traversaro, 2023-03-15:
> Does Dart engine not support the use of ray related plug-ins at present

No, this is exactly the issue tracked at https://github.com/gazebosim/gazebo-classic/issues/911 .

huqin-RM, 2023-03-15:
> 

At present, can anyone successfully use ray under the dart engine?

traversaro, 2023-03-15:
No, because the relevant code is not implemented, see https://github.com/gazebosim/gazebo-classic/issues/911 . I will close this issue, if you have more thing to add feel free to write them in https://github.com/gazebosim/gazebo-classic/issues/911, thanks!

huqin-RM, 2023-03-15:
> No, because the relevant code is not implemented, see #911 . I will close this issue, if you have more thing to add feel free to write them in #911, thanks!

Ok