gazebosim/gazebo-classic #3308: Model gets teleported with all links to origin

srediske, 2023-03-22:
## Environment
* OS Version: Ubuntu 20.04
* Binary gazebo11_11.11.0

## Description

We observed a strange behavior in the simulation: The model is sometimes teleported to the origin with all its links and cannot be used afterwards, so it has to be deleted from the sim. It only occurs with the world solver ([ODE WorldStep](https://opende.sourceforge.net/docs/group__world.html#ga11a3871bb135dc53f12feb6e116fc27e))

This Video shows the problem:

https://user-images.githubusercontent.com/25820310/226826259-c7e81622-a1b7-466f-8d44-61cfd03907a9.mp4

What stands out here is the long-lasting jumping of the contact points. In this model, a nested model is placed with the `<include>` tag, which sits loosely without a joint. If the nested model is deactivated, it looks like this:

https://user-images.githubusercontent.com/25820310/226826280-0cfa5df1-9490-4dc4-9f36-11eb099f7ce3.mp4

It's stable just after spawn. But the problem still exists, sooner or later the model will teleport (broken) to origin, at least after interaction with the model. We also tested some models from the official gazebo repo, e.g. the coke can:

https://user-images.githubusercontent.com/25820310/226826433-f4ecbce0-3458-4f7a-a28c-96dd5879d920.mp4

And what is going on with the contacts after another model is spawned? The contacts from the coke cans look stable and the terminal doesn't spam any annoying ODE errors (`ODE Message 3: LCP internal error, s <= 0 (s=...)`), but when i spawn another model they suddenly jump around. Also, note that the coke can had jumping contacts right before it teleported to the origin:

https://user-images.githubusercontent.com/25820310/226826387-077be96b-10ec-4b5d-b3cf-eba5522db602.mp4

Problems with the same symptom but probably different cause:

1. [Hardware interface as supposed cause](https://answers.gazebosim.org/question/27233/robot-teleporting-to-origin-with-broken-joints/?sort=oldest)
2. revolute2 joint type as supposed cause: #2239
3. [Same behavior seen with the demo joint types model](https://github.com/gazebosim/gazebo-classic/issues/2401#issuecomment-616314447) from #2401 

## Steps to reproduce

1. Start world file:
```
<?xml version="1.0" ?>
<sdf version="1.7">
  <world name="default">
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <include>
      <uri>model://sun</uri>
    </include>

    <physics name="world_solver" type="ode">
      <ode>
        <solver>
          <type>world</type>
          <sor>1.35</sor>
        </solver>
        <constraints>
          <cfm>0.0005</cfm>
        </constraints>
      </ode>
    </physics>

  </world>
</sdf>
```
2. Spawn a simple model with oscillating contacts, e.g. the coke can from the [gazebo_models](https://github.com/osrf/gazebo_models) repo
3. Move/Rotate the model to create more contact changes until it moves to the origin

## Output

traversaro, 2023-03-22:
"Model gets teleported to the origin" is the high level behaviour that correspond to the fact that at some points a NaN emerged in simulation, probably as a result of a numerically unstable simulation. Probably it would be easier to debug if you log the actual model state.

srediske, 2023-03-22:
Thanks for your reply!

Here is a log file with a coke_can: [state.log](https://github.com/gazebosim/gazebo-classic/files/11037969/state.log). 

I can't find any NaN and it seems to me that the only useful part of this log is /physics/contacts and /pose/info, or did I misunderstand logging the model state?