gazebosim/gazebo-classic #3310: Incorrect wide angle camera lens flare occlusion after camera movement

mogumbo, 2023-03-29:
## Environment
* Ubuntu 20.04
* gazebo11 d3584008b47c745c1732a6cd07af68726cd900fa

## Description
* Expected behavior: After moving a robot with a wide angle camera and then triggering an image, lens flare should be occluded or not occluded as expected.
* Actual behavior: Sometimes lens flare is not occluded by objects, and sometimes lens flare does not appear when there is no object to occlude it.

## Steps to reproduce
After moving your robot this call to `occlusionScale` will sometimes give an incorrect result: https://github.com/gazebosim/gazebo-classic/blob/gazebo11/gazebo/rendering/LensFlare.cc#L314

If you inspect the view matrix just before that call to occlusion scale, it will not be up to date. Move your robot, trigger an image, and then trigger a second image. You should see that the view matrix changes between image triggers. Here's some code for inspecting the view matrix:
```
            const Ogre::Matrix4& m = this->dataPtr->wideAngleDummyCamera->OgreCamera()->getViewMatrix();
            std::cerr << m[0][0] << "  " << m[0][1] << "  " << m[0][2] << "  " << m[0][3] << std::endl;
            std::cerr << m[1][0] << "  " << m[1][1] << "  " << m[1][2] << "  " << m[1][3] << std::endl;
            std::cerr << m[2][0] << "  " << m[2][1] << "  " << m[2][2] << "  " << m[2][3] << std::endl;
            std::cerr << m[3][0] << "  " << m[3][1] << "  " << m[3][2] << "  " << m[3][3] << std::endl;
```
Digging a little deeper, you can see that in `Scene::FirstContact` https://github.com/gazebosim/gazebo-classic/blob/gazebo11/gazebo/rendering/Scene.cc#L1217 the `origin` and `dir` used to compute occlusions are incorrect because the view matrix is incorrect.

## Output

scpeters, 2023-03-30:
I added an additional model with a pair of wide-angle cameras updated at 1 Hz to a test world in https://github.com/gazebosim/gazebo-classic/commit/a43e4a3f2df9f68d467b9325397a39dce94272a3 and was able to reproduce the issue:

https://user-images.githubusercontent.com/3650755/228989801-091c99f3-d7ab-4691-a2d3-3618c49fa0f5.mov

scpeters, 2023-05-19:
there is a fix in review in https://github.com/gazebosim/gazebo-classic/pull/3325

PR #3325 (2023-05-25): Fix wide-angle lens flare occlusion lag
Fixes #3310 

This forces an update of the wide angle dummy camera, which updates the Ogre camera's view matrix. 

---

One way to check that this fixes the bug, is to use [this](https://github.com/audrow/gazebo-classic/blob/audrow/debug-lens-flare/worlds/lensflare_wideangle_cam.world) world file. Open it in Gazebo, and then move the only camera (`wide_angle_cameras_occluded_higher`) from it's starting height (`z=21`) to `z=11`, where it will have a wall between it and the light source. You can then move it back to `z=21`.

You should see lens flare when [... cut]
closed 2023-05-25 by scpeters